LibreCAD/LibreCAD #697: Cannot compile master or 2.1.0 alpha with Qt4

vlachoudis, 2016-04-12:
I don't know if it is the Qt4 or something else but I cannot compile the current master or the 2.1.0-alpha
I get the following errors during compilation
```
cd libraries/ && make -f Makefile 
/home/bnv/git/LibreCAD/t/LibreCAD-master/libraries
make[1]: Entering directory '/home/bnv/git/LibreCAD/t/LibreCAD-master/libraries'
cd libdxfrw/ && make -f Makefile 
/home/bnv/git/LibreCAD/t/LibreCAD-master/libraries/libdxfrw
make[2]: Entering directory '/home/bnv/git/LibreCAD/t/LibreCAD-master/libraries/libdxfrw'
g++ -c -pipe -g -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -std=gnu++98 -Wno-deprecated -O2 -Wall -W -fPIC -D_REENTRANT -DQT_NO_DEBUG -I/usr/lib64/qt4/mkspecs/linux-g++ -I. -I/usr/include -I../../generated/lib/libdxfrw/moc -o ../../generated/lib/libdxfrw/obj/libdxfrw.o src/libdxfrw.cpp
In file included from src/libdxfrw.h:17:0,
                 from src/libdxfrw.cpp:14:
src/drw_entities.h:103:17: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
  DRW_Entity() = default;
                 ^
src/drw_entities.h:104:26: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
  virtual ~DRW_Entity() = default;
                          ^
src/drw_entities.h:134:31: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  enum DRW::ETYPE eType = DRW::UNKNOWN;     /*!< enum: entity type, code 0 */
                               ^
src/drw_entities.h:135:24: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  duint32 handle = DRW::NoHandle;            /*!< entity identifier, code 5 */
                        ^
src/drw_entities.h:137:30: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  duint32 parentHandle = DRW::NoHandle;      /*!< Soft-pointer ID/handle to owner BLOCK_RECORD object, code 330 */
                              ^
src/drw_entities.h:138:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  DRW::Space space = DRW::ModelSpace;          /*!< space indicator, code 67*/
                          ^
src/drw_entities.h:139:21: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  UTF8STRING layer = "0";          /*!< layer name, code 8 */
                     ^
src/drw_entities.h:140:24: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  UTF8STRING lineType = "BYLAYER";       /*!< line type, code 6 */
                        ^
src/drw_entities.h:141:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  duint32 material = DRW::MaterialByLayer;          /*!< hard pointer id to material object, code 347 */
                          ^
src/drw_entities.h:142:19: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int color = DRW::ColorByLayer;                 /*!< entity color, code 62 */
                   ^
src/drw_entities.h:143:53: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  enum DRW_LW_Conv::lineWidth lWeight = DRW_LW_Conv::widthByLayer; /*!< entity lineweight, code 370 */
                                                     ^
src/drw_entities.h:144:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  double ltypeScale = 1.0;         /*!< linetype scale, code 48 */
                      ^
src/drw_entities.h:145:17: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  bool visible = true;              /*!< entity visibility, code 60 */
                 ^
src/drw_entities.h:146:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int numProxyGraph = 0;         /*!< Number of bytes in proxy graphics, code 92 */
                      ^
src/drw_entities.h:148:17: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int color24 = -1;               /*!< 24-bit color, code 420 */
                 ^
src/drw_entities.h:150:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int transparency = DRW::Opaque;          /*!< transparency, code 440 */
                          ^
src/drw_entities.h:151:23: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int plotStyle = DRW::DefaultPlotStyle;             /*!< hard pointer id to plot style object, code 390 */
                       ^
src/drw_entities.h:152:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  DRW::ShadowMode shadow = DRW::CastAndReceieveShadows;    /*!< shadow mode, code 284 */
                                ^
src/drw_entities.h:153:23: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  bool haveExtrusion = false;        /*!< set to true if the entity have extrusion*/
                       ^
src/drw_entities.h:154:14: error: 'shared_ptr' is not a member of 'std'
[... 566 lines cut ...]
  ^
src/libdxfrw.cpp:2513:28: error: expected primary-expression before '>' token
  std::shared_ptr<DRW_Vertex> v = std::make_shared<DRW_Vertex>();
                            ^
src/libdxfrw.cpp:2513:30: error: 'v' was not declared in this scope
  std::shared_ptr<DRW_Vertex> v = std::make_shared<DRW_Vertex>();
                              ^
src/libdxfrw.cpp:2513:34: error: 'make_shared' is not a member of 'std'
  std::shared_ptr<DRW_Vertex> v = std::make_shared<DRW_Vertex>();
                                  ^
src/libdxfrw.cpp:2513:61: error: expected primary-expression before '>' token
  std::shared_ptr<DRW_Vertex> v = std::make_shared<DRW_Vertex>();
                                                             ^
src/libdxfrw.cpp:2513:63: error: expected primary-expression before ')' token
  std::shared_ptr<DRW_Vertex> v = std::make_shared<DRW_Vertex>();
                                                               ^
Makefile:256: recipe for target '../../generated/lib/libdxfrw/obj/libdxfrw.o' failed
make[2]: *** [../../generated/lib/libdxfrw/obj/libdxfrw.o] Error 1
make[2]: Leaving directory '/home/bnv/git/LibreCAD/t/LibreCAD-master/libraries/libdxfrw'
Makefile:40: recipe for target 'sub-libdxfrw-make_default' failed
make[1]: *** [sub-libdxfrw-make_default] Error 2
make[1]: Leaving directory '/home/bnv/git/LibreCAD/t/LibreCAD-master/libraries'
Makefile:42: recipe for target 'sub-libraries-make_default-ordered' failed
make: *** [sub-libraries-make_default-ordered] Error 2
```

r-a-v-a-s, 2016-04-12:
Try editing libdxfrw.pro

```
QMAKE_CXXFLAGS_DEBUG -= -std=c++11
QMAKE_CXXFLAGS -= -std=c++11
```

Either remove that or change `-=` to `+=`

vlachoudis, 2016-04-12:
Great! by converting it them to += it compiled hapilly

vlachoudis, 2016-04-13:
I was too fast. It worked on Ubuntu 15.10 but not on my FC23, where I get the following messages.

```
cd libraries/ && make -f Makefile 
/home/bnv/git/LibreCAD/LibreCAD-master/libraries
make[1]: Entering directory '/home/bnv/git/LibreCAD/LibreCAD-master/libraries'
cd libdxfrw/ && make -f Makefile 
/home/bnv/git/LibreCAD/LibreCAD-master/libraries/libdxfrw
make[2]: Entering directory '/home/bnv/git/LibreCAD/LibreCAD-master/libraries/libdxfrw'
g++ -c -pipe -g -std=c++11 -std=c++11 -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic -std=gnu++98 -Wno-deprecated -O2 -Wall -W -fPIC -D_REENTRANT -DQT_NO_DEBUG -I/usr/lib64/qt4/mkspecs/linux-g++ -I. -I/usr/include -I../../generated/lib/libdxfrw/moc -o ../../generated/lib/libdxfrw/obj/libdxfrw.o src/libdxfrw.cpp
In file included from src/libdxfrw.h:17:0,
                 from src/libdxfrw.cpp:14:
src/drw_entities.h:103:17: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
  DRW_Entity() = default;
                 ^
src/drw_entities.h:104:26: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
  virtual ~DRW_Entity() = default;
                          ^
src/drw_entities.h:134:31: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  enum DRW::ETYPE eType = DRW::UNKNOWN;     /*!< enum: entity type, code 0 */
                               ^
src/drw_entities.h:135:24: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  duint32 handle = DRW::NoHandle;            /*!< entity identifier, code 5 */
                        ^
src/drw_entities.h:137:30: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  duint32 parentHandle = DRW::NoHandle;      /*!< Soft-pointer ID/handle to owner BLOCK_RECORD object, code 330 */
                              ^
src/drw_entities.h:138:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  DRW::Space space = DRW::ModelSpace;          /*!< space indicator, code 67*/
                          ^
src/drw_entities.h:139:21: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  UTF8STRING layer = "0";          /*!< layer name, code 8 */
                     ^
src/drw_entities.h:140:24: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  UTF8STRING lineType = "BYLAYER";       /*!< line type, code 6 */
                        ^
src/drw_entities.h:141:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  duint32 material = DRW::MaterialByLayer;          /*!< hard pointer id to material object, code 347 */
                          ^
src/drw_entities.h:142:19: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int color = DRW::ColorByLayer;                 /*!< entity color, code 62 */
                   ^
src/drw_entities.h:143:53: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  enum DRW_LW_Conv::lineWidth lWeight = DRW_LW_Conv::widthByLayer; /*!< entity lineweight, code 370 */
                                                     ^
src/drw_entities.h:144:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  double ltypeScale = 1.0;         /*!< linetype scale, code 48 */
                      ^
src/drw_entities.h:145:17: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  bool visible = true;              /*!< entity visibility, code 60 */
                 ^
src/drw_entities.h:146:22: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int numProxyGraph = 0;         /*!< Number of bytes in proxy graphics, code 92 */
                      ^
src/drw_entities.h:148:17: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int color24 = -1;               /*!< 24-bit color, code 420 */
                 ^
src/drw_entities.h:150:26: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int transparency = DRW::Opaque;          /*!< transparency, code 440 */
                          ^
src/drw_entities.h:151:23: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  int plotStyle = DRW::DefaultPlotStyle;             /*!< hard pointer id to plot style object, code 390 */
                       ^
src/drw_entities.h:152:32: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  DRW::ShadowMode shadow = DRW::CastAndReceieveShadows;    /*!< shadow mode, code 284 */
                                ^
src/drw_entities.h:153:23: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
  bool haveExtrusion = false;        /*!< set to true if the entity have extrusion*/
                       ^
src/drw_entities.h:154:14: error: 'shared_ptr' is not a member of 'std'
[... 566 lines cut ...]
  ^
src/libdxfrw.cpp:2513:28: error: expected primary-expression before '>' token
  std::shared_ptr<DRW_Vertex> v = std::make_shared<DRW_Vertex>();
[... cut ...]
Makefile:256: recipe for target '../../generated/lib/libdxfrw/obj/libdxfrw.o' failed
make[2]: *** [../../generated/lib/libdxfrw/obj/libdxfrw.o] Error 1
make[2]: Leaving directory '/home/bnv/git/LibreCAD/LibreCAD-master/libraries/libdxfrw'
Makefile:40: recipe for target 'sub-libdxfrw-make_default' failed
make[1]: *** [sub-libdxfrw-make_default] Error 2
make[1]: Leaving directory '/home/bnv/git/LibreCAD/LibreCAD-master/libraries'
Makefile:42: recipe for target 'sub-libraries-make_default-ordered' failed
make: *** [sub-libraries-make_default-ordered] Error 2
```

r-a-v-a-s, 2016-04-13:
What is FC23? What is the exact version of Qt?

vlachoudis, 2016-04-14:
Fedora Core 23, and I have the qt-devel-4.8.7-12.fc23.x86_64

r-a-v-a-s, 2016-04-14:
Are the compilers identical?

vlachoudis, 2016-04-14:
No. ubuntu has
gcc version 5.2.1 20151010 (Ubuntu 5.2.1-22ubuntu2)
with qt4-dev 4:4.8.6+git64-g5dc8b2b+dfsg-3~ubuntu8

And fedora has
gcc version 5.3.1 20151207 (Red Hat 5.3.1-2) (GCC)
with qt version 4.8.7-12.fc23

vlachoudis, 2016-04-14:
WIth the new changes I get the following error on Ubuntu as well
```
In file included from ui/forms/qg_dlgoptionsgeneral.h:29:0,
                 from ui/qg_dialogfactory.cpp:72:
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h: In member function 'void Ui_QG_DlgOptionsGeneral::setupUi(QDialog*)':
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:194:29: error: 'class QComboBox' has no member named 'setCurrentText'
         cbHighlightedColor->setCurrentText(QString::fromUtf8("#739373"));
                             ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:213:26: error: 'class QComboBox' has no member named 'setCurrentText'
         cbSelectedColor->setCurrentText(QString::fromUtf8("#A54747"));
                          ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:232:22: error: 'class QComboBox' has no member named 'setCurrentText'
         cbGridColor->setCurrentText(QString::fromUtf8("#736F6E"));
                      ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:263:28: error: 'class QComboBox' has no member named 'setCurrentText'
         cbBackgroundColor->setCurrentText(QString::fromUtf8("#000000"));
                            ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:276:26: error: 'class QComboBox' has no member named 'setCurrentText'
         cbMetaGridColor->setCurrentText(QString::fromUtf8("#404040"));
                          ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:289:29: error: 'class QComboBox' has no member named 'setCurrentText'
         cbStartHandleColor->setCurrentText(QString::fromUtf8("#00FFFF"));
                             ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:307:24: error: 'class QComboBox' has no member named 'setCurrentText'
         cbHandleColor->setCurrentText(QString::fromUtf8("#0000FF"));
                        ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:331:27: error: 'class QComboBox' has no member named 'setCurrentText'
         cbEndHandleColor->setCurrentText(QString::fromUtf8("#0000FF"));
                           ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:348:24: error: 'class QComboBox' has no member named 'setCurrentText'
         cb_snap_color->setCurrentText(QString::fromUtf8("#FFC200"));
                        ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:453:28: error: 'class QComboBox' has no member named 'setCurrentText'
         cb_layerselection->setCurrentText(QString::fromUtf8("#CCFFCC"));
                            ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:531:35: error: 'class QComboBox' has no member named 'setCurrentText'
         indicator_lines_combobox->setCurrentText(QString::fromUtf8("Crosshair"));
                                   ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:550:35: error: 'class QComboBox' has no member named 'setCurrentText'
         indicator_shape_combobox->setCurrentText(QString::fromUtf8("Circle"));
                                   ^
```

r-a-v-a-s, 2016-04-14:
That's weird... All I did was uncheck "translatable" for the current text in the ui designer.
It doesn't make sense for Qt4 to create a ui_ file that uses a function it doesn't have.
I can try reverting those changes...

r-a-v-a-s, 2016-04-14:
Try with that commit.

vlachoudis, 2016-04-14:
Ubuntu fails with
```
In file included from ui/forms/qg_dlgoptionsgeneral.h:29:0,
                 from ui/forms/qg_dlgoptionsgeneral.cpp:26:
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h: In member function 'void Ui_QG_DlgOptionsGeneral::retranslateUi(QDialog*)':
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:898:29: error: 'class QComboBox' has no member named 'setCurrentText'
         cbHighlightedColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#739373", 0, QApplication::UnicodeUTF8));
                             ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:900:26: error: 'class QComboBox' has no member named 'setCurrentText'
         cbSelectedColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#A54747", 0, QApplication::UnicodeUTF8));
                          ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:902:22: error: 'class QComboBox' has no member named 'setCurrentText'
         cbGridColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#736F6E", 0, QApplication::UnicodeUTF8));
                      ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:905:28: error: 'class QComboBox' has no member named 'setCurrentText'
         cbBackgroundColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#000000", 0, QApplication::UnicodeUTF8));
                            ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:906:26: error: 'class QComboBox' has no member named 'setCurrentText'
         cbMetaGridColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#404040", 0, QApplication::UnicodeUTF8));
                          ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:907:29: error: 'class QComboBox' has no member named 'setCurrentText'
         cbStartHandleColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#00FFFF", 0, QApplication::UnicodeUTF8));
                             ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:909:24: error: 'class QComboBox' has no member named 'setCurrentText'
         cbHandleColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#0000FF", 0, QApplication::UnicodeUTF8));
                        ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:912:27: error: 'class QComboBox' has no member named 'setCurrentText'
         cbEndHandleColor->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#0000FF", 0, QApplication::UnicodeUTF8));
                           ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:916:24: error: 'class QComboBox' has no member named 'setCurrentText'
         cb_snap_color->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#FFC200", 0, QApplication::UnicodeUTF8));
                        ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:928:28: error: 'class QComboBox' has no member named 'setCurrentText'
         cb_layerselection->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "#CCFFCC", 0, QApplication::UnicodeUTF8));
                            ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:936:35: error: 'class QComboBox' has no member named 'setCurrentText'
         indicator_lines_combobox->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "Crosshair", 0, QApplication::UnicodeUTF8));
                                   ^
../../generated/librecad/ui/ui_qg_dlgoptionsgeneral.h:938:35: error: 'class QComboBox' has no member named 'setCurrentText'
         indicator_shape_combobox->setCurrentText(QApplication::translate("QG_DlgOptionsGeneral", "Circle", 0, QApplication::UnicodeUTF8));
                                   ^
Makefile:75161: recipe for target '../../generated/librecad/obj/qg_dlgoptionsgeneral.o' failed
make[2]: *** [../../generated/librecad/obj/qg_dlgoptionsgeneral.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

And Fedora with
```
Makefile:301: recipe for target '../../generated/lib/libdxfrw/obj/drw_entities.o' failed
make[2]: *** [../../generated/lib/libdxfrw/obj/drw_entities.o] Error 1
Makefile:275: recipe for target '../../generated/lib/libdxfrw/obj/libdwgr.o' failed
make[2]: *** [../../generated/lib/libdxfrw/obj/libdwgr.o] Error 1
Makefile:256: recipe for target '../../generated/lib/libdxfrw/obj/libdxfrw.o' failed
make[2]: *** [../../generated/lib/libdxfrw/obj/libdxfrw.o] Error 1
src/jwwdoc.cpp: In member function 'CDataList JWWBlockList::GetBlockList(unsigned int)':
src/jwwdoc.cpp:1762:9: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
  return {};
         ^
src/jwwdoc.cpp: In member function 'CDataEnko JWWBlockList::GetCDataEnko(int, int)':
src/jwwdoc.cpp:1774:9: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
  return {};
         ^
src/jwwdoc.cpp: In member function 'CDataMoji JWWBlockList::GetCDataMoji(int, int)':
src/jwwdoc.cpp:1781:9: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
  return {};
         ^
src/jwwdoc.cpp: In member function 'CDataSen JWWBlockList::GetCDataSen(int, int)':
src/jwwdoc.cpp:1788:9: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
  return {};
         ^
src/jwwdoc.cpp: In member function 'CDataSolid JWWBlockList::GetCDataSolid(int, int)':
src/jwwdoc.cpp:1795:9: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
  return {};
[... cut ...]
src/jwwdoc.cpp:1809:9: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
  return {};
         ^
src/jwwdoc.cpp: In member function 'CDataBlock JWWBlockList::GetCDataBlock(int, int)':
src/jwwdoc.cpp:1936:9: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
  return {};
         ^
make[2]: Leaving directory '/home/bnv/git/LibreCAD/libraries/libdxfrw'
Makefile:40: recipe for target 'sub-libdxfrw-make_default' failed
make[1]: *** [sub-libdxfrw-make_default] Error 2
```

r-a-v-a-s, 2016-04-14:
With Ubuntu it's again creating a ui_ file that uses a Qt5 function...
The only thing I can think of is that you somehow have the Qt5 uic
http://doc.qt.io/qt-4.8/uic.html

With Fedora... maybe run clean.sh or make distclean... I don't know.

vlachoudis, 2016-04-14:
Sorry @r-a-v-a-s but it didn't fix none of the two. 
BTW on ubuntu I don't have any Qt5 and on Fedora I removed for testing.

r-a-v-a-s, 2016-04-14:
The only thing I can think of is that you need to upgrade to 4.8.7 on Ubuntu.
It still doesn't make sense though...

Maybe dxli can help with the c++11 issues... those weren't my changes.

r-a-v-a-s, 2016-04-15:
Manually removing the currentText blocks in qg_dlgoptionsgeneral.ui
would probably solve that issue... but that seems suboptimal.

Either I'm missing something, the ubuntu Qt package isn't setup correctly, 
or Qt hasn't considered this problem.

r-a-v-a-s, 2016-04-17:
> My guess would be that UIC doesn't necessarily know the classes, i.e. it has to be able to deal with custom widgets, or widgets derived from standard widgets ("promote widget" functionality in designer).

> So when it encounters a property value assignment, it might simply assume that the setter for the property is set + property name as this is true for all of Qt's built-in properties.

http://www.qtcentre.org/threads/65811-qt4-uic-uses-qt5-function

That sure makes supporting Qt4 much more inconvenient...

Try building with the latest commit.

vlachoudis, 2016-04-17:
Thanks @r-a-v-a-s the new master compiles in Ubuntu. On Fedora 23 I still have the problem with the c++

vlachoudis, 2016-05-03:
@dxli In Fedora 23 with g++ 5.3.1, to compile drw_entities.cpp I managed to get rid of some error messages with the following changes:
1. `= default` to {}  I understand that is going backwards vs c++11
2. `<DRW_Variant>>`  to `<DRW_Variant> >`  with a space between the two >>
3. And all `std::shared_ptr` to use `std::tr1::shared_ptr`
4. the same for `make_shared`
I've tried also the option -std=c++0x  but didn't work

r-a-v-a-s, 2016-05-14:
Is there any reason not to use that fix?

The master branch will become 2.2.0-alpha before the end of the month;
and that will mark the end of Qt4 support.
The 2.1 branch (once created) will maintain support for Qt4.

vlachoudis, 2016-05-14:
I didn't manage to get rid of all messages. Also the std::tr1:: trick is deprecated.

dxli, 2016-05-14:
@vlachoudis 

did you try to build from scratch? remove the old folder, and start with a new git clone.

all the hacking you are trying suggests you do NOT have c++11 with gcc-5.3.1.

read the full compiler line, make sure you have an option **-std=c++11**, please do not use <del>-std=c++0x</del>

also try to specify compiler options from qmake command line:

http://stackoverflow.com/questions/23050998/qmake-assigning-variables-in-command-line

since the default "qmake -r;make" should build LC, I'm wondering whether it's some of your local configurations, like bashrc, etc.

r-a-v-a-s, 2016-09-07:
https://apps.fedoraproject.org/packages/librecad/builds/
There is a 2.1.0 package.
I think dxli is correct.

closed 2016-09-07 by r-a-v-a-s