LibreCAD/LibreCAD #698: Menu too heavy

vlachoudis, 2016-04-14:
I personally (maybe it is only me) find the menu too heavy and scary :)
I would suggest the following changes:
- Move the "Options" as a submenu inside the "File" or in "Edit"
- Move the "Widgets" as a submenu of "Windows"
- Move the "Help" on the far right to be visually separated
- Move the "Plugins" which is less used before the "Window"
- I would even remove the "Block" and "Layer" as long as they appear as a Dockwidget on the right
  I would put them back only if someone selects not to use the "Dockwidgets"
- In an even more extreme way I would do the same with all toolbars that are displayed.

r-a-v-a-s, 2016-04-14:
1. I dislike looking for options in menus... one of my favorite changes is the toplevel Options menu.

2. It's too inefficient for the Widget's menu to be a submenu.

3. I'm not aware of a way to move the Help menu as you have described.

4. The location of the Plugins menu doesn' really matter to me as long as it's top level.

5. I've never used the Block or Layer menus; therefore, I wouldn't be against moving or removing them.

vlachoudis, 2016-04-14:
1 & 2 My approach would be, to have immediately visible what is frequently/commonly used. What is less frequent could be "hidden" as a submenu, to avoid cluttering the display. That was the reasoning behind Options and Widgets, which you might need them when you set up librecad or a drawing, but during the editing they are not so useful.
3. I don't know how to use Qt, so I cannot be of much help :)
4. The order for me should be from left to right reflecting higher to lower importance. This is typically how the eye (for western countries) will search the tool bar.

r-a-v-a-s, 2016-04-14:
I understand what you are saying, but it's subjective.
I only use the File, Options, Widgets and Help menus.

I might add the option to exclude certain menus.

vlachoudis, 2016-04-14:
You are right it is subjective. I personally don't use the menu at all :) and with all new screens which are good for playing games but not for working the vertical space is precious. I would even prefer to hide it and when pressing Alt to show the menu temporarily.

JonnyTech, 2016-04-15:
3\. Does this help?: http://stackoverflow.com/questions/8726677/aligning-qmenubar-items-add-some-on-left-and-some-on-right-side

```
QMenuBar *bar = new QMenuBar(ui->menuBar);

QMenu *menu = new QMenu("Test menu", bar);
bar->addMenu(menu);

QAction *action = new QAction("Test action", bar);
bar->addAction(action);

ui->menuBar->setCornerWidget(menubar);
```

[image]

The trick is to use **setCornerWidget**

r-a-v-a-s, 2016-04-15:
Thanks. I guess the truth is I didn't research it because I find it... weird.

This is my solution:
[image]

- draw -> tools
- window -> drawings
- removed snap, block and layers menu
- added modify, select, info and dimensions to the tools menu

We will see if anyone misses those menus.

vlachoudis, 2016-04-16:
It is not my program, I simply wanted to help in optimising from the perspective of the end user (but maybe it is only me). As you changed it is much nicer, I would prefer however a different order.

File, Edit, View, Drawing, Tools, Plugins, Options, Widget, Windows, Help

I would put back the window on top level, when you edit multiple documents you need to switch easily. Tabbed windows are nice but eat up space.

r-a-v-a-s, 2016-04-16:
I meant to say Drawings = Window.
Simply a name change there.
I also considered 'Documents'... 
either is clearer than 'Window'.

closed 2016-05-10 by r-a-v-a-s