LibreCAD/LibreCAD #700: inconsistent inclusion of library files

r-a-v-a-s, 2016-04-22:
I'm noticing on openSUSE that the library browser contains everything from
https://github.com/LibreCAD/LibreCAD/tree/master/librecad/support/library

On OS X the library browser is empty, and on Windows it's just a couple folders.

The difference is from the postprocess scripts.

What exactly do we want to include?

Would it make more sense to maintain this folder in the Resources repository,
and offer it as an additional download?
That would simplify the postprocess scripts,
and provide a clean library browser for people who would prefer that.
However, it would produce confusion 
when someone opens a file that used one of those blocks.

fa201, 2023-08-08:
Is this still an issue ?

lordofbikes, 2023-08-09:
This was fixed with 168597a12170b5378ea09c71eb7d5a89afcbcf0f.

Anyhow, we already have https://dokuwiki.librecad.org/doku.php/community:partlibs where additional part libraries are listed for download.

I just created https://dokuwiki.librecad.org/doku.php/community:fonts for additional fonts.
I consider to move some fonts, and possibly some libraries too, to the wiki to reduce the installer package sizes.

closed 2023-08-09 by lordofbikes