LibreCAD/LibreCAD #704: commit id in the about window on ubuntu

r-a-v-a-s, 2016-04-27:
@maqifrnswa I've noticed when people who are using your ppa report something the commit id is "2.1.0".
It's supposed to show the 7 digit SHA, and it does on Windows, OS X and openSUSE.

Version: 2.1.0-alpha
Commit: 2.1.0
Compiler: GNU GCC 4.8.4
Compiled on: Apr 25 2016
Qt Version: 5.2.1
Boost Version: 1.54.0

I believe this is the associated line in src.pro
`SCMREVISION=$$system([ "$(which git)x" != "x" -a -d ../../.git ] && echo "$(git describe --tags)" || echo "$${SCMREVISION}")`

and in qc_applicationwindow.cpp
`tr("Commit: %1").arg(QString(XSTR(QC_SCMREVISION)).right(7))`

I don't fully understand the line in src.pro
but maybe it would make sense to start using `git rev-parse --short HEAD`.

r-a-v-a-s, 2016-05-13:
`QMAKE_POST_LINK = "$$_PRO_FILE_PWD_/../../scripts/postprocess-win.bat" $$SCMREVISION`

Is `$$SCMREVISION` being passed to `postprocess-win.bat`?
If yes: where is it used?

lordofbikes, 2016-05-14:
Yes, that's the meaning of this line.
`$$SCMREVISION` is passed on to the NSIS installer, where it is used to insert it in a registry key on setup.

lordofbikes, 2016-05-14:
Concerning your original question, 2.1.0 is the default value of `$$SCMREVISION` which is set on line 11 of src.pro. 
The line you quoted try to locate git and calls `git describe --tags` if it was found. If this fails, `echo "$${SCMREVISION}"` keeps the default value for the info dialog. For the other OS it's similar.
This is the desired behaviour, because not everyone who build LC himself has git installed.
So there is no problem in LC build files, there is an issue on the build system for the ppa binaries, where git seems not to be found.
The advantage of `git describe --tags` is that it references the latest tag of the current branch and is more descriptive than `git rev-parse --short HEAD`.

closed 2016-05-16 by r-a-v-a-s