LibreCAD/LibreCAD #705: 'Save' does not save a customized scale and position in 'preview'

gerardbm, 2016-04-30:
First of all, thanks for this awesome software.

Everytime I switch to the Preview Impression I have to select the scale again (1:100, instead of 1:113.514) and I have to move the drawing in a certain position.

This happens because of autocenter and autoscale. These two functions remove a saved customized impression. Also, these two functions are redundant because there are already two buttons at bottom for these cases: 'center' and 'adjust'.

Perhaps some people want autoscale and autocenter. A consensus solution would be a checkbox to allow us enable/disable these functions.

Regards.

r-a-v-a-s, 2016-05-02:
https://github.com/LibreCAD/LibreCAD/issues/592

Cantar4, 2016-05-03:
#592 in mind, solution or not at the horizon, it would be good to get a reminder 
of the scale (be it a manual 1.5:1 or an autoscale psychedelic 1.43025:1),
systematically printed at an extreme  corner of the page.

gerardbm, 2019-01-31:
This issue is fixed after the pull request 1050. It's an important improvement I've been waiting for ~3 years. With this, LibreCAD is 1000 times better IMHO, so I can save the drawings, close them, reopen the files 5 months later, modify something and print it to PDF without having to compose the layout again. Thanks, @xanderdin!

Cantar4, 2019-01-31:
100% with you but there is no more sailor at the whelm !!
 Is Lord of bikes able to release us a nightly built to this request?
--Cantar4

xanderdin, 2019-01-31:
I'm glad if my contribution makes you happy. But dear @gerardbm, please keep in mind that the 'Fixed' checkbox value is not saved in drawings. It is saved to global LibreCAD configuration. This means that if you uncheck the 'Fixed' checkbox during preview of another drawing and forget to check it back, then opening yet another drawing, which is supposed to be fixed, could still break the scale. That is how it is made, and I believe that is because dxf file cannot keep that 'fixed' value.

gerardbm, 2019-01-31:
That's good to know, @xanderdin. Anyway, it's a good improvement, I only have to check the 'Fixed' checkbox to save every file with its scale. What would happen if I opened the file in another installation of LibreCAD. Would it work?

Off topic: By the way, what would be intereseting is to have infinite previews with different layers, so I can have a model with 100 layers, a preview with 30 layers, another preview with another composition of layers, and so on... (According to your comment, I suppose DXF is not prepared for this behaviour).

xanderdin, 2019-01-31:
Yes @gerardbm, it should work in another installation of LibreCAD because the scale itself is saved in DXF file. With another installation I would recommend, before opening your saved drawings, run LibreCAD with default drawing ('unnamed document 1') first, switch to print preview and make sure the 'fixed' checkbox is checked. After that, it should be safe to open any drawings with their saved scales.

Concerning infinite previews with different layers, I'm barely enough acquainted with DXF internals, so I just don't know if it is possible.

lordofbikes, 2019-02-01:
@Cantar4, nightly builds for Windows are build automatically, you find them on [sourceforge](https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/), the merge was on Jan 5, so the latest build from Jan 6, LibreCAD-Installer-2.2.0-rc1-104-g10289432, contains this fix.
For other OS we sadly have no nightly builds yet, you can try to build from source.

@gerardbm, basically DXF format supports views. But they are meant to store different points of views or sections of the drawing, not varying sets of layers.
Iirc there were some requests in past, can't find them yet, but there is nobody working on this feature.
Concerning the layers, maybe a clever naming system could help. The layer list has a filter, where you can use patterns to filter the layers, then you can use the buttons to show or hide all filtered layers without affecting the other layers.

closed 2019-02-01 by lordofbikes