LibreCAD/LibreCAD #706: lc_actiongroupmanager.h missing #include <QMap.h>

DL7BJ, 2016-05-05:
On Ubuntu 14.04 64 Bit build stops with QMap does not name a type. 
After adding #include <QMap.h> to lc_actiongroupmanager.h build is ok.

r-a-v-a-s, 2016-05-05:
Odd that it built on 3 different platforms, and travis ci didn't have a problem with it.
What are you using to compile?

DL7BJ, 2016-05-05:
I use this How-To: http://librecad.org/cms/home/from-source/linux.html

gcc (Ubuntu 4.8.4-2ubuntu1~14.04.1) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.

g++ (Ubuntu 4.8.4-2ubuntu1~14.04.1) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.

	linux-vdso.so.1 =>  (0x00007ffdd5af1000)
	libQtSvg.so.4 => /usr/lib/x86_64-linux-gnu/libQtSvg.so.4 (0x00007f5ebf28a000)
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f5ebe5d7000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f5ebe0f2000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f5ebded4000)
	libmuparser.so.2 => /usr/lib/x86_64-linux-gnu/libmuparser.so.2 (0x00007f5ebdc7e000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f5ebd97a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f5ebd674000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f5ebd45e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5ebd099000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f5ebce80000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f5ebcc44000)
	libaudio.so.2 => /usr/lib/x86_64-linux-gnu/libaudio.so.2 (0x00007f5ebca2c000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f5ebc724000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f5ebc4fe000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f5ebc25b000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f5ebc00a000)
	libSM.so.6 => /usr/lib/x86_64-linux-gnu/libSM.so.6 (0x00007f5ebbe02000)
	libICE.so.6 => /usr/lib/x86_64-linux-gnu/libICE.so.6 (0x00007f5ebbbe6000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f5ebb9d6000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f5ebb7cc000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f5ebb5ba000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f5ebb285000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5ebb081000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f5ebae79000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5ebf4e1000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f5ebac4f000)
	libXt.so.6 => /usr/lib/x86_64-linux-gnu/libXt.so.6 (0x00007f5eba9e9000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f5eba7e5000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f5eba5a7000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f5eba39f000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f5eba19a000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f5eb9f7b000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f5eb9d75000)

I have no problem to build LibreCad after adding the include to lc_actiongroupmanager.h with qmake/make.

r-a-v-a-s, 2016-05-05:
Maybe it's because you are using an older version of gcc.
I did add the patch... thanks!

closed 2016-05-05 by r-a-v-a-s

r-a-v-a-s, 2016-05-05:
Part of the problem is the instructions you used,
which are for Qt4. Those really need to be removed from the website.

https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

DL7BJ, 2016-05-05:
Okay, thanks for the info! 

I use a LTS version of Ubuntu, can't really change my development system so often, use it for embedded devices, Lazarus & FLTK. The gcc is the actual version in the repository.