LibreCAD/LibreCAD #708: Text Width Factor

CivilGuy, 2016-05-07:
I'm having trouble with using text with a width factor slightly less than 1.

I'd like to use the standard AutoCAD simplex font, but with a width factor of 0.95.  It'll just look better, I think.  I've gone through the process of Modify Properties for a few Text objects and changed the width factor to 0.95.  The change is drawn okay, but doesn't seem to be saved properly.

The next time the drawing is opened, the width factor is changed back to 1.

r-a-v-a-s, 2016-05-16:
:white_check_mark: Confirmed

CivilGuy, 2017-11-14:
I've found the bug, I think.  In file rs_filterdxfrw.cpp, RS_FilterDXFRW::writeText(RS_Text* t), there's no line that says to copy the value of the relative width from the RS_Text instance to the DRW_Text instance, so the value is written out from the DRW_Text instance as the default 1.0.

I'll start a pull request.

fa201, 2020-04-13:
@lordofbikes 
Tested with a multitext with a size of 2 changed to 1.8 since there is no width factor. The file is saved and re-opened and the size is still 1.8.
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

pdr0663, 2023-04-08:
There is a further bug here...

I tried applying a width factor of 0.8 to the "iso" font and it produced a very dodgy rendering. I note also that 'Iso3098" and  "iso3098b" also don't render properly (check "0" character).