LibreCAD/LibreCAD #709: File menu box in the French version not scaling properly

JPSimardMG2, 2016-05-09:


r-a-v-a-s, 2016-05-09:
That's not enough information. Add a screenshot if you can't explain it further.

r-a-v-a-s, 2016-05-09:
Also, please include the version of LibreCAD, the version of Qt and your operating system.

JPSimardMG2, 2016-05-09:
When i click on the menu bar for Fichier it covers the whole screen. Uninstalled and reinstalled issue persist.

[image]
[image]

r-a-v-a-s, 2016-05-09:
It's because the file path is not truncated.
I have been thinking about putting the recent files in a sub-menu.

JPSimardMG2, 2016-05-09:
LibreCAD-Installer-2.0.10.exe
QT 5.4.1
Win10

r-a-v-a-s, 2016-05-09:
Thanks 😎  I'll work on it.

JPSimardMG2, 2016-05-09:
No problem glad to help you out.

closed 2016-05-10 by r-a-v-a-s