LibreCAD/LibreCAD #711: Auto Zoom should ignore origin

lordofbikes, 2016-05-12:
As discussed on IRC, there is an issue with auto zoom.

There is a drawing about 300,000 by 300,000 units, with several entities, around X 3,500,00 Y 5,400,000, which is typical for land surveying. There are no entities around the origin. 
Using auto zoom shows the origin at the bottom of the window and a little spot representing the drawing on the window top. Zooming manually into the drawing is working fine.

This was seen by the user in LC 2.0.9 on Debian and I could reproduce it with 2.1.0-alpha on Ubuntu.

lordofbikes, 2016-05-19:
I've checked both, master and 2.0 with the file from IRC, both work fine.
Thanks for the fix.

r-a-v-a-s, 2016-05-26:
Unfortunately this fix adds much more prominent problems:
http://forum.librecad.org/Auto-zoom-troubles-with-2-1-beta-4-td5713760.html
https://github.com/LibreCAD/LibreCAD/issues/723

r-a-v-a-s, 2016-05-26:
More unfortunate is that dxli can't work on this now...
I'll revert this patch later today unless someone thinks they can solve the other problems.

fa201, 2020-04-13:
@lordofbikes 
I checked with a 100x100 rectangle at 2 000 000, 3 000 0000 and auto zoom have the rectangle fit on the drawing area.

rdorsch, 2020-04-13:
Works much better now with 2.1.3

Would it make sense to run autozoom automatically, after a file is loaded?