LibreCAD/LibreCAD #714: Unable to copy hatch area

Equoda, 2016-05-16:
Hatch doesn't copy by "Edit -> Copy | Edit -> Paste".

To reproduce:

1. Create draft
2. Create "Rectangle" shape
3. Select created shape
4. Do "Tools -> Hatch"
5. Select shape with hatch
6. "Edit -> Copy"
7. "Edit -> Paste"

Result: no hatch in pasted shape.

Librecad build info:

Version: 2.1.0-beta
Tag: 2.1.0-beta-8-g9723828
Compiler: GNU GCC 4.8.4
Compiled on: May 16 2016
Qt Version: 4.8.6
Boost Version: 1.54.0

Notes:

Draft is created from template. Here is: 
[draft-A4-book-mm.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/265915/draft-A4-book-mm.dxf.zip)

Equoda, 2016-05-17:
The same apply to "Move" shape with hatch: hatch disappears.

Equoda, 2016-05-17:
The same apply to "Create Block" with shape containing hatch: hatch disappears. Sometimes, it appears again when "explode" such block with invisible hatch.

Equoda, 2016-05-17:
The same apply to moving hatch to another layer. In this case "Undo" could bring it back.

Equoda, 2016-05-17:
The same apply to changing hatch properties with "Properties" tool.

Equoda, 2016-05-23:
There are debug messages while running with `--debug 6` switch (pasting a rectangle 40x40 filled with "ansi31" hatch):

> RS_Undo::startUndoCycle
> RS_LayerList::activate notify: 0 begin
> RS_LayerList::activate end
> RS_EntityContainer::detach: autoDel: 1
> RS_EntityContainer::clone: ori autoDel: 1
> RS_EntityContainer::clone: clone autoDel: 1
> RS_EntityContainer::detach: autoDel: 1
> RS_Hatch::update
> RS_Hatch::update: contour has 1 loops
> RS_Hatch::update
> RS_EntityContainer::optimizeContours
> RS_EntityContainer::optimizeContours: OK
> RS_Hatch::update: invalid contour in hatch found
> RS_Hatch::update
> RS_Hatch::update: contour has 1 loops
> RS_Hatch::update
> RS_EntityContainer::optimizeContours
> RS_EntityContainer::optimizeContours: OK
> RS_Hatch::update: invalid contour in hatch found
> RS_Undo::addUndoable(): begin
> RS_Undo::addUndoable(): end
> RS_EntityContainer::detach: autoDel: 1
> RS_EntityContainer::calculateBorders
> RS_EntityContainer::calculateBorders: size 1: 40.000000,40.000000
> RS_EntityCotnainer::calculateBorders: size: 40.000000,40.000000
> RS_EntityContainer::calculateBorders
> RS_EntityContainer::calculateBorders: size 1: 40.000000,40.000000
> RS_EntityCotnainer::calculateBorders: size: 40.000000,40.000000
> RS_Undo::addUndoable(): begin
> RS_Undo::addUndoable(): end
> RS_Undo::addUndoCycle
> RS_Undo::addUndoCycle: ok
> RS_ActionInterface::finish
> RS_ActionInterface::finish: OK
> RS_EventHandler::cleanUp
> RS_EventHandler::cleanUp: OK
> QG_GraphicView::mouseReleaseEvent: OK

Is there any way to trace execution with more details not running Librecad in debugger?

Equoda, 2016-05-24:
This patch partitially solve the problem for me, but I'm not sure this doesn't cause any other regressions. The code of ```RS_EntityContainer::optimizeContours()``` function some kind of messy. The pasted hatch does not preserve some attributes: color of lines "by layer", for example.

[hatch_bug.diff.zip](https://github.com/LibreCAD/LibreCAD/files/280404/hatch_bug.diff.zip)

```diff

diff --git a/librecad/src/lib/engine/rs_entitycontainer.cpp b/librecad/src/lib/engine/rs_entitycontainer.cpp
index d7cb579..e0c87b8 100644
--- a/librecad/src/lib/engine/rs_entitycontainer.cpp
+++ b/librecad/src/lib/engine/rs_entitycontainer.cpp
@@ -1565,24 +1565,34 @@ bool RS_EntityContainer::optimizeContours() {
     /** connect entities **/
     const QString errMsg=QObject::tr("Hatch failed due to a gap=%1 between (%2, %3) and (%4, %5)");
 
-    while(count()>0){
+    while(count()>0) {
         double dist(0.);
         RS_Vector&& vpTmp=getNearestEndpoint(vpEnd,&dist,&next);
         if(dist>1e-8) {
-            if(vpEnd.squaredTo(vpStart)<1e-8){
+            if(vpEnd.squaredTo(vpStart) < 1e-8) {
                 RS_Entity* e2=entityAt(0);
                 tmp.addEntity(e2->clone());
                 vpStart=e2->getStartpoint();
                 vpEnd=e2->getEndpoint();
                 removeEntity(e2);
                 continue;
-			}
-			QG_DIALOGFACTORY->commandMessage(
-						errMsg.arg(dist).arg(vpTmp.x).arg(vpTmp.y).arg(vpEnd.x).arg(vpEnd.y)
-						);
-            closed=false;
+            }
+            else {
+                QG_DIALOGFACTORY->commandMessage(
+                            errMsg.arg(dist).arg(vpTmp.x).arg(vpTmp.y).arg(vpEnd.x).arg(vpEnd.y)
+                            );
+                RS_DEBUG->print(RS_Debug::D_ERROR, "RS_EntityContainer::optimizeContours: hatch failed due to a gap");
+                closed=false;
+                break;
+            }
         }
-		if(next && closed){ 			//workaround if next is nullptr
+        if(!next) { 	    //workaround if next is nullptr
+//      	    std::cout<<"RS_EntityContainer::optimizeContours: next is nullptr" <<std::endl;
+            RS_DEBUG->print("RS_EntityContainer::optimizeContours: next is nullptr");
+//			closed=false;	//workaround if next is nullptr
+            break;			//workaround if next is nullptr
+        } 					//workaround if next is nullptr
+        if(closed) {
             next->setProcessed(true);
             RS_Entity* eTmp = next->clone();
             if(vpEnd.squaredTo(eTmp->getStartpoint())>vpEnd.squaredTo(eTmp->getEndpoint()))
@@ -1590,20 +1600,15 @@ bool RS_EntityContainer::optimizeContours() {
             vpEnd=eTmp->getEndpoint();
             tmp.addEntity(eTmp);
         	removeEntity(next);
-		} else { 			//workaround if next is nullptr
-//      	    std::cout<<"RS_EntityContainer::optimizeContours: next is nullptr" <<std::endl;
-
-			closed=false;	//workaround if next is nullptr
-			break;			//workaround if next is nullptr
-		} 					//workaround if next is nullptr
+        }
     }
 //    DEBUG_HEADER
-    if(vpEnd.valid && vpEnd.squaredTo(vpStart)>1e-8) {
-		if(closed)
-			QG_DIALOGFACTORY->commandMessage(errMsg.arg(vpEnd.distanceTo(vpStart))
-											 .arg(vpStart.x).arg(vpStart.y).arg(vpEnd.x).arg(vpEnd.y));
-        closed=false;
-    }
+//    if(vpEnd.valid && vpEnd.squaredTo(vpStart) > 1e-8) {
+//		QG_DIALOGFACTORY->commandMessage(errMsg.arg(vpEnd.distanceTo(vpStart))
+//											 .arg(vpStart.x).arg(vpStart.y).arg(vpEnd.x).arg(vpEnd.y));
+//        RS_DEBUG->print("RS_EntityContainer::optimizeContours: hatch failed due to a gap");
+//        closed=false;
+//    }
 //    std::cout<<"RS_EntityContainer::optimizeContours: 5"<<std::endl;
 
 
@@ -1614,7 +1619,12 @@ bool RS_EntityContainer::optimizeContours() {
     }
 //    std::cout<<"RS_EntityContainer::optimizeContours: 6"<<std::endl;
 
-    RS_DEBUG->print("RS_EntityContainer::optimizeContours: OK");
+    if(closed) {
+        RS_DEBUG->print("RS_EntityContainer::optimizeContours: OK");
+    }
+    else {
+        RS_DEBUG->print("RS_EntityContainer::optimizeContours: bad");
+    }
 //    std::cout<<"RS_EntityContainer::optimizeContours: end: count()="<<count()<<std::endl;
 //    std::cout<<"RS_EntityContainer::optimizeContours: closed="<<closed<<std::endl;
     return closed;

```

r-a-v-a-s, 2016-05-24:
You might have to be your own hero on this... >_<
There are no active team members that work in this area.

If you become confident in your solution, the please submit a pull request.

https://github.com/LibreCAD/LibreCAD/wiki/Git-and-GitHub
https://github.com/LibreCAD/LibreCAD/wiki/Becoming-a-developer
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

PR #738 (2016-06-07): Committer: Equoda <Equoda@users.noreply.github.com>
fix: moving/pasting hatch bug

Solve the problem with hatch moving/pasting cause hatch disappears.
This is a quick solution.
There are some minor issues to solve: hatch doesn't appear on the same layer as original after moving.

Closes #714
commit c6b154a7 (2016-06-07): Committer: Equoda <Equoda@users.noreply.github.com>

fix: moving/pasting hatch bug

Solve the problem with hatch moving/pasting cause hatch disappears.
This is a quick solution.
There are some minor issues to solve: hatch doesn't appear on the same layer as original after moving.

Closes #714
closed 2016-06-07 by r-a-v-a-s