LibreCAD/LibreCAD #715: Can't move a number of blocks

Equoda, 2016-05-16:
Can't "Move" a number of blocks, copied from another draft.

Steps to reproduce:

1. Create new draft
2. Create a shape ("Rectangle")
3. Create a block from shape ("Create Block -> $block_name")
4. Repeat 2. and 3. (I created 3 blocks)
5. Select all, "Edit -> Copy"
6. Create new draft
7. "Edit -> Paste"
8. Select all, "Move (delete original)"

Result: some shapes disappears.

Librecad info:

Version: 2.1.0-beta
Tag: 2.1.0-beta-8-g9723828
Compiler: GNU GCC 4.8.4
Compiled on: May 16 2016
Qt Version: 4.8.6
Boost Version: 1.54.0

Drafts are created from templates. Here is an example of draft with shapes: 
[blocks.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/265934/blocks.dxf.zip)

Equoda, 2016-05-31:
Here is the patch which solve this problem for me. Actual for librecad developer version (build: Version: 2.1.0-beta-70-g9522a79, Compiled on: May 31 2016).

[librecad.blocks-bug.2016.05.31.diff.zip](https://github.com/LibreCAD/LibreCAD/files/291852/librecad.blocks-bug.2016.05.31.diff.zip)

Most changes are in the ```./librecad/src/lib/modification/rs_modification.cpp```. Function ```paste``` has been rewritten.

Made a few tests without errors.

r-a-v-a-s, 2016-05-31:
Please fork using the button at the top right of this page.
Then clone your new fork in the folder of your choice (e.g. repositories):
`git clone https://github.com/Equoda/LibreCAD.git`

Keep your master branch synced
https://help.github.com/articles/configuring-a-remote-for-a-fork/
https://help.github.com/articles/syncing-a-fork/

And work from branches you create:
`git checkout your_branch`

Then when the time comes you can submit a pull request:
https://help.github.com/articles/using-pull-requests/

r-a-v-a-s, 2016-05-31:
More info at: https://github.com/LibreCAD/LibreCAD/wiki/Git-and-GitHub

I use https://git-cola.github.io/ to look at differences and create commits.

Equoda, 2016-05-31:
OK. I will check this later on...

PR #737 (2016-06-07): Committer: Equoda <Equoda@users.noreply.github.com>
fix: blocks moving/pasting bug

Solve the problem with copy-paste or move blocks bug cause some blocks disappears.
Most changes are in ```librecad/src/lib/modification/rs_modification.cpp```.
```RS_Modification::paste``` function has been rewritten.

Closes #715
commit d6d5b037 (2016-06-07): Committer: Equoda <Equoda@users.noreply.github.com>

fix: blocks moving/pasting bug

Solve the problem with copy-paste or move blocks bug cause some blocks disappears.
Most changes are in ```librecad/src/lib/modification/rs_modification.cpp```.
```RS_Modification::paste``` function has been rewritten.

Closes #715
closed 2016-06-07 by r-a-v-a-s