LibreCAD/LibreCAD #716: Removed blocks names are persistent somewhere

Equoda, 2016-05-16:
Block names are persistent somewhere in memory when corresponding blocks have been removed from "Block List". Unable to create new block with the same name as deleted. Closing and reopening draft solve this problem.

Steps to reproduce:

1. Create new draft.
2. Create a shape ("rectangle").
3. Select shape, create block ("Create Block -> 'block_1'").
4. Remove 'block_1'.
5. Create a shape ("rectangle").
6. Select shape, try to create block ("Create Block -> 'block_1' -> 'Could not name block. A block named "block_1" already exists.'").
7. Save ("block_names.dxf"), close draft.
8. Open draft ("block_names.dxf").
9. Select shape, create block ("Create Block -> 'block_1'").

Result: successful.

lordofbikes, 2016-05-16:
This is a known limitation, not exactly a bug. It is by design, caused by the undo function.
There are several tickets on sf.net, see [bug ticket 463](https://sourceforge.net/p/librecad/bugs/463/).

I'll close the sf.net ticket to make this issue the new place for further discussions.

Equoda, 2016-05-17:
Yes, I think removed blocks remaining for "undo" should be renamed or deleted by user choice. It's very annoying to give a new name for every updated block in composite draft like: "melting_chamber_1", "melting_chamber_2", etc. Maybe, blocks should be renamed after deletion such as: "melting_chamber_rem#0001" and restored after "undo" with this name, so user could decide.

shawncurry, 2019-05-24:
Armin @lordofbikes ,

Couldn't these sort of things be handled by making the setUndoState() virtual, allowing specialized handling for certain entities as needed?  If a block was able to rename itself when it changes undo state, it seems like the old name could then be recycled. 

Just thinking out loud.  If you want to sound out some ideas, I'd be interested.

shawncurry, 2019-05-24:
Better still, introduce a new, virtual function, to be invoked by setUndoState() at the appropriate time.

shawncurry, 2019-05-25:
Well I tried it out, and I think I understand a little better the design limitation you were referring to.  To "undo" a block could mean, to add vs remove it from the block list, but what about renaming a block?  Seems this cannot currently be undone.  Perhaps an "RS_UndoableAction" could be a solution to these sorts of limitations.

pseudo-code:

class RS_UndoableAction : public RS_Undoable {
  virtual void doAction() = 0;
  virtual void undoAction() = 0;
  virtual void undoStateChanged(bool undone) {
    if (undone)
      undoAction();
    else
      doAction();
  }
}

class RS_RenameBlockAction : public RS_UndoableAction {
  // . . . 
  void doAction() {
    block.setName(newName);
  }
  void undoAction() {
    block.setName(oldName);
  }
}

shawncurry, 2019-05-26:
I have successfully completed a significant refactoring and redesign of the RS_Undoable classes.  I introduced the classes discussed above, and a couple more.  I also successfully used the new classes to implement undo/redo the naming of a block.  Here is a brief overview my my commit notes.

- Extracted RS_Undo implementation from RS_Document; now it has an RS_Undo, instead of being one
- Introduce RS_UndoManager as an implementation of RS_Undo, also as notification center for undo events
- Introduce RS_UndoListener, RS_Document now uses this interface to be notified of changes (instead of performing them)
- Introduce RS_UndoableAction as a solution for undoing actions other than add/remove entity
- Introduce RS_BlockRenameAction as an example of an undoable action

The problem discussed by @Equoda still exists at the moment, but I'm close to having that fixed as well.  This is an unintended behavior owing to a bit of a design shortcut; likely taken in part because of the previous lack of a listener/notification center system for undo events.  Undone blocks will now actually be removed from the block list, and tranferred to a separate "undone" list.

There will be a lot more work to be done with the new undo capabilities.  In the previous design, the only type of action that could be undone was adding or removing an Entity, which does cover a lot of ground.  

Using the new classes, any type of state changes can now be easily undone as well, like renaming, freezing, visibility changes, etc.  One needs only to implement the RS_UndoableAction interface, and then they will be able to do was previously thought impossible: the doUndoableAction(). haha

Symbian9, 2023-02-27:
Latest LibreCAD 2.2.0 release still has issue with persisting "removed blocks".

Any progress on improving Undo capabilities?

- https://github.com/LibreCAD/LibreCAD/issues/1109

u2fly, 2025-04-21:
- https://github.com/LibreCAD/LibreCAD/issues/716#issuecomment-496026382
>The problem discussed by @Equoda still exists at the moment, but I'm close to having that fixed as well.

@shawncurry, Was it fixed in https://github.com/hypertherm/LibreCAD fork?

@dxli @sand1024