LibreCAD/LibreCAD #718: Cannot Ungroup entities & blocks

sparxz, 2016-05-16:
In ver 2.0.10.

I have entities that are in same layer somehow. I can select them but cannot ungroup them to delete the items I want to separate.
I have read manual & I have searched term: "Librecad Ungroup entities & blocks " and nothing comes up.
Something I can do easily in Scribus and Libreoffice.

Am I missing somthing ?

If I cannot get beyond this then I will have to abandon Librecad

lordofbikes, 2016-05-16:
Try this:
- go to **Modify** menu
- select **Explode** tool
- the mouse cursor has become a cross, then select the block(s) you want to separate
- press **ENTER**
- you're done, now you can select single entities of the former block(s) to delete them

sparxz, 2016-05-16:
Worked. Thanks so much :-))

I hope they change the drop-down menu to something typically CAD/DTP familiar like "Ungroup" or
 better still "ungroup objects" /  "Ungroup Elements"

I would have associated "Explode" with the typical long-standing technical drawing term "Exploded view".

lordofbikes, 2016-05-17:
You're welcome.

As I know CAD, **Explode** is the standard wording for that function [e.g. AutoCAD](https://knowledge.autodesk.com/support/autocad-lt/learn-explore/caas/CloudHelp/cloudhelp/2016/ENU/AutoCAD-LT/files/GUID-E98BCEF4-DED6-48A6-87EB-10FE87188083-htm.html). I think it's hard to compare CAD and DTP, imho they have not much in common.

closed 2016-05-17 by lordofbikes

sparxz, 2016-05-18:
I use the amazing Opensource 'Scribus' program which is a kind of LibreCad of DTP world ;-)
It handles layers in the same way, and locking vectored elements in a similar way,

lordofbikes, 2016-05-18:
There is nothing wrong with Scribus or DTP or your request to change the wording. But for CAD, the Explode command and most others are established and it's not recommend to change this.