LibreCAD/LibreCAD #719: Printing does not work on OSX

realrolfje, 2016-05-16:
Printing a document on Librecad 2.0.14 on OSX 10.11.4 does not work.
- Prints are scaled up to 5 times the original size (scalings are all set to 1)
- PDF preview from the print dialog does not work reliably, or doesn't even open
- export to PDF produces portrait PDF with just the bottom left part of the drawing (due to scale)
- export to SVG produces strange underscaled results with missing lines
- Opening the DXF in LibreOffice Draw results in missing objects (parts of blocks)

Exporting to PNG is the only thing that sort of works, but the scaling and resolution settings are unclear, and I can't get a 1-to-1 scale 600dpi png out of it.

I have found no way to print out my drawing at a 1-to-1 scale. (I wish I found this bug before finishing my drawing)

r-a-v-a-s, 2016-05-19:
There is no 2.0.14.
Can you check the information in the About window?

Regarding export to PDF, check the paper size (e.g. A4) in Current Drawing Preferences.
https://github.com/LibreCAD/LibreCAD/issues/553

Try using Export as MakerCAM SVG
You could then import it into InkScape for other export options.

fa201, 2021-12-27:
Is this issue still valid ?