LibreCAD/LibreCAD #722: Something broke in Ver 2.1.0-beta compiled on May 18 2016

FRKgh, 2016-05-18:
PPAs:
deb http://ppa.launchpad.net/librecad-dev/librecad-daily/ubuntu trusty main
deb-src http://ppa.launchpad.net/librecad-dev/librecad-daily/ubuntu trusty main

Latest download:
Version: 2.1.0-beta
SCM Revision: 2.1.0
Compiler: GNU GCC 4.8.4
Compiled on: May 18 2016
Qt Version: 5.2.1
Boost Version: 1.54.0

Issues:

Issue 1
**File** > **Recent Files** > Open an existing file (works fine)
**Auto Zoom** > Everything disappears
**Print Preview** > Nothing visible (original fixed scale of 1:48 changed to 1:1e+09)

Issue 2
Start LibreCAD with default blank document
**Print Preview** on blank document
LibreCAD freezes. Have to **Force Quit**

OS: LinuxMint 17.3 Rosa

Is there something wrong at my end? I have had a completely malfunctioning revision before; fortunately, it got fixed with subsequent daily updates.

How does one roll-back to a previous revision?

Thank you.

r-a-v-a-s, 2016-05-18:
Make two issues; and have their titles be the most concise description of the issue (e.g. "program freezes on print preview").

I can't reproduce issue 1.
I can reproduce issue 2.

r-a-v-a-s, 2016-05-18:
I simplified this issue so you can just make one more for the other.

It looks like the program is getting stuck in a loop... debug:
```
RS_ActionInterface::RS_ActionInterface: Setting up action: "Print Preview"
RS_ActionInterface::RS_ActionInterface: Setting up action: "Print Preview": OK
QG_DialogFactory::requestOptions
QG_DialogFactory::requestOptions: OK
  adding listeners
  showing MDI window
RS_VariableDict::addVariable()
```
Then it just continuously produces `RS_VariableDict::addVariable()`

lordofbikes, 2016-05-18:
The removed issue 1 could be a duplicate of #711, maybe @FRKgh can check this.

r-a-v-a-s, 2016-05-18:
It was a recursive error.
I don't know what changed to cause this,
nor do I fully understand the section of code;
however, the change does avoid the freezing.

Hopefully someone can review this.

FRKgh, 2016-05-19:
Version: 2.1.0-beta
Compiler: GNU GCC 4.8.4
Compiled on: May 19 2016
Qt Version: 5.2.1
Boost Version: 1.54.0

Freezing resolved.

**Print Preview** and **Auto Zoom** resolved, BUT, only for drawings created in this build.

**Print Preview** and **Auto Zoom** still do not work for drawings that used to work in builds two revisions ago.

closed 2016-05-19 by FRKgh