LibreCAD/LibreCAD #723: Auto Zoom malfunctioning (ver 2.1.0-beta; May 18 2016 compile)

FRKgh, 2016-05-18:
Latest download:
Version: 2.1.0-beta
SCM Revision: 2.1.0
Compiler: GNU GCC 4.8.4
Compiled on: May 18 2016
Qt Version: 5.2.1
Boost Version: 1.54.0

OS: LinuxMint 17.3 Rosa

Have been using LibreCAD for months without any issue with Auto Zoom. Use it several times every day. This latest build has it not functioning properly.

My existing drawing switches to the following grid size in **Grid Status** when Auto Zoom is clicked on:
1.50995e+08 / -1
Grid size is supposed to be 12 / 36 when correctly zoomed.

Is that an issue with LibreCAD or Ubuntu?

How does one roll-back to a previous revision?

Thank you.

FRKgh, 2016-05-18:
Please note that this happens only with drawings created in previous builds of LibreCAD.

If a new drawing is created with the latest build of LibreCAD, then the **Auto Zoom** function works properly.

r-a-v-a-s, 2016-05-19:
> How does one roll-back to a previous revision?

You could use the stable ppa.
Otherwise you will need to wait, or learn how to build from source.

r-a-v-a-s, 2016-05-19:
If you decide to build from source use:
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source
https://github.com/LibreCAD/LibreCAD/wiki/Becoming-a-developer
Not the official website!

You can use git checkout https://git-scm.com/docs/git-checkout
to switch to a specific commit.

FRKgh, 2016-05-19:
Stable ppa resolved the Auto Zoom issue.

However, it does crash sometimes - quits program unexpectedly. That was the reason I had switched to daily updates, which are very stable; not a single instance of crashing.

The latest builds have excellent toolbar interface, which the present stable ppa doesn't. Hopefully, those toolbars will be incorporated in the next stable version.

r-a-v-a-s, 2016-05-19:
2.1.0 will be released in a few days... as you have noticed it's light years ahead of 2.0.10 ;-]
Although you will once again have this problem unless it is fixed...

The only commit that seems related (and might have caused this) is: https://github.com/LibreCAD/LibreCAD/commit/add5df5b4c765b756b8587ad46c9af5262f3d4ff

r-a-v-a-s, 2016-05-28:
It has been reported as fixed by another tester.
Wait until it updates then you could try the next "daily" 
https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-daily

r-a-v-a-s, 2016-05-29:
There appears to still be problems with drawings made using https://github.com/LibreCAD/LibreCAD/commit/add5df5b4c765b756b8587ad46c9af5262f3d4ff

It is requested that anyone interested also review:
http://forum.librecad.org/Auto-zoom-troubles-with-2-1-beta-4-td5713760.html

r-a-v-a-s, 2016-05-31:
@FRKgh Can you try the latest daily? I added another fix.

FRKgh, 2016-06-01:
Version: 2.1.0-beta
Compiler: GNU GCC 4.8.4
Compiled on: Jun  1 2016
Qt Version: 5.2.1
Boost Version: 1.54.0

Tested with the latest version shown above. Works perfectly with couple of drawings I tested. 

Coincidentally, the other issue of rotated PDF export, with Ledger/Tabloid, also has started working.

Thank you very much!

closed 2016-06-04 by r-a-v-a-s