LibreCAD/LibreCAD #725: "Linear zeros -> remove right" removes integer part zeros in dimensions

Equoda, 2016-05-19:
When using a stripping zeros option from "Options -> Current drawing preferences -> Dimensions -> Linear zeros [remove right]" along with "Options -> Current drawing preferences -> Dimensions -> Linear precision [0]" integer part zeros of this dimension are removed.

Steps to reproduce:

1. Create a new draft
2. Create a shape ("Rectangle [40x40]")
3. Create a dimension mark for 2. ("Dimension -> Horizontal"). Created "Dimension" view is: 40.000
4. Open "Options -> Current drawing preferences -> Dimensions", select "Linear precision [0]" and "Linear zeros [remove right]", apply

Result: created "Dimension" view is: 4, not 40 as expected.

Librecad info:

Version: 2.1.0-beta-22-gebcbf53
Compiler: GNU GCC 4.8.4
Compiled on: May 19 2016
Qt Version: 4.8.6
Boost Version: 1.54.0

Sample draft: 
[size_zeros.dxf.zip](https://github.com/LibreCAD/LibreCAD/files/272416/size_zeros.dxf.zip)

r-a-v-a-s, 2016-05-22:
:white_check_mark: Confirmed.

r-a-v-a-s, 2016-05-22:
The problem is possibly in [`RS_Dimension::stripZerosLinear`][1]
Trying to read it makes me sleepy... :open_mouth: 

[1]:http://wiki.librecad.org/lc2doc/class_r_s___dimension.html#a49007ba90b699dd6c6a562c8522f0499

Equoda, 2016-05-31:
This patch should close the bug. Actual for git developer version (build: Version: 2.1.0-beta-70-g9522a79, Compiled on: May 31 2016).

[librecad.dimensions-bug.2016.05.31.diff.zip](https://github.com/LibreCAD/LibreCAD/files/292106/librecad.dimensions-bug.2016.05.31.diff.zip)

There are may remain troubles with angular dimensions. I haven't tested all cases.

closed 2016-06-07 by r-a-v-a-s

r-a-v-a-s, 2016-09-20:
Do you want to add this to the 2.1 branch?

Equoda, 2016-09-22:
What should I do? Make equivalent changes to 2.1? OK, I will check

r-a-v-a-s, 2016-09-22:
Yep. For small commits without conflict you can use https://git-scm.com/docs/git-cherry-pick
I mostly do it manually;
and I keep a secondary clone to avoid switching branches and rebuilding.

PR #739 (2016-06-07): Committer: Equoda <Equoda@users.noreply.github.com>
fix: wrong dimension format when removing trailing zeroes

Solve the removing integer part of dimension problem. Decimal sparator was ignored.
Changes are in ```librecad/src/lib/engine/rs_dimension.cpp```. ```RS_Dimension::stripZerosLinear``` fuction has been rewritten.

Closes #725
commit d7302543 (2016-06-07): Committer: Equoda <Equoda@users.noreply.github.com>

fix: wrong dimension format when removing trailing zeroes

Solve the removing integer part of dimension problem. Decimal sparator was ignored.
Changes are in ```librecad/src/lib/engine/rs_dimension.cpp```. ```RS_Dimension::stripZerosLinear``` fuction has been rewritten.

Closes #725