LibreCAD/LibreCAD #726: cannot open dwg file

jirislaby, 2016-05-19:
I am trying to open this dwg:
http://www.fi.muni.cz/~xslaby/sklad/M500/syrovice_k.dwg

librecad writes after a while:
RS_FilterDXFRW::fileImport: reading DWG file: OK
RS_DialogFactory::instance()
RS_DialogFactory::instance(): OK
RS_DialogFactory::getFactoryObject
RS_DialogFactory::getFactoryObject: returning factory object
QG_DialogFactory::commandMessage
QG_DialogFactory::commandMessage: OK
RS_DialogFactory::instance()
RS_DialogFactory::instance(): OK
RS_DialogFactory::getFactoryObject
RS_DialogFactory::getFactoryObject: returning factory object
QG_DialogFactory::commandMessage
QG_DialogFactory::commandMessage: OK
RS_FilterDXFRW::printDwgError: DRW::BAD_READ_TABLES
Cannot open DWG file '/tmp/mozilla_xslaby0/M500/syrovice_k.dwg'.

With dwg, I also got:
http://www.fi.muni.cz/~xslaby/sklad/M500/syrovice_k.dgn
(I don't know if that is useful)

Full "librecad -d 6 output":
http://www.fi.muni.cz/~xslaby/sklad/M500/debug.txt

jirislaby, 2016-05-20:
I am on openSUSE with librecad 2.0.10.

FelixNotTheCat, 2017-01-27:
DRW_Block_Record::parseDwg() extracts the compressed .dwg block records literally bitwise and maps them to their open/LC representation. However, for some block records in that drawing this reconstruction goes wrong, making the final integrity check fail. @jirislaby do you know in what version the .dwg was saved (e.g. 2004, 2007)?

jirislaby, 2017-01-27:
In technical report for that file I see: 
> Digital processing was performed by the graphic program MICROSTATION PowerDraft V8i with output to *.dgn and conversion to *.dwg.
So I assume both were created by PowerDraft...

FelixNotTheCat, 2017-06-08:
for info: have stopped debugging, sorry. the issue culminates in the LibreCad application specifying the various binary formats that it is committed to handle and how, so that the process of opening files implements the relevant criteria and validation. as it stands, the given .dwg file is simply not opened on the basis of its binaries not being structured as LibreCad-expected of a 2007 dwg (if i remember correctly).

lordofbikes, 2017-06-08:
No reason to apologize @FelixNotTheCat, many thanks for your efforts to try to battle this. 
And thanks for sharing your results.