LibreCAD/LibreCAD #727: librecad is unusably slow with much text

jirislaby, 2016-05-19:
When I open this dxf:
http://www.fi.muni.cz/~xslaby/sklad/test.dxf
librecad cannot be used reliably. It is way too slow when moving and operating. I assume it is because of the plenty of text there.

Note that I deleted most of the drawings. If I do the opposite: leave drawings and delete text, it works almost fine. I.e. it is only a bit slower.

jirislaby, 2016-05-20:
I am on openSUSE with librecad 2.0.10.

jirislaby, 2016-05-20:
I built it with -pg -mfentry and the profiler says (cropped to top 10):
Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  ms/call  ms/call  name    
 12.54      0.38     0.38 253630493     0.00     0.00  RS_Entity::getLayer(bool) const
 10.37      0.69     0.31  5654719     0.00     0.00  RS_Entity::getPen(bool) const
  3.68      0.80     0.11  2411481     0.00     0.00  RS_Line::draw(RS_Painter*, RS_GraphicView*, double&)
  2.68      0.88     0.08 12087377     0.00     0.00  RS_Entity::RS_Entity(RS_EntityContainer*)
  2.51      0.95     0.08 88603338     0.00     0.00  RS_Entity::isUndone() const
  2.01      1.01     0.06  6495686     0.00     0.00  RS_GraphicView::drawEntityPlain(RS_Painter*, RS_Entity*, double&)
  2.01      1.07     0.06  2766696     0.00     0.00  RS_GraphicView::drawEntity(RS_Painter*, RS_Entity*, double&)
  1.84      1.13     0.06 19154983     0.00     0.00  RS_Entity::isVisible() const
  1.67      1.18     0.05  3369092     0.00     0.00  void std::vector<RS_Vector, std::allocator<RS_Vector> >::_M_emplace_back_aux<RS_Vector const&>(RS_Vector const&)
  1.67      1.23     0.05  2770510     0.00     0.00  RS_PainterQt::setPen(RS_Pen const&)

Whole profile at:
http://www.fi.muni.cz/~xslaby/sklad/profile.txt

r-a-v-a-s, 2016-05-20:
Does enabling draft mode help?

---

It seems to go from `RS_GraphicView::drawEntity` to `RS_GraphicView::drawEntityPlain`
but then I don't understand the call to `e->draw(painter, this, patternOffset);`
Does it go to `RS_EntityContainer::draw` and then `RS_GraphicView::drawEntity`?

It's another case of the code seemingly going it circles... 
and all the overloaded functions make things less clear.

jirislaby, 2016-05-21:
With draft mode, it is usable, yes.

If you want to see the calling sequence, take a look at the full dump and look for "Call graph". There you can see the call graph (sequence).

r-a-v-a-s, 2016-05-21:
If the code is written in a way that I can't follow the flow of execution using Qt Creator,
then I don't really have any interest in working on it.
There are too many other more interesting things I can not get paid to do... ;-]

r-a-v-a-s, 2016-06-07:
Since `RS_Text` inherits `RS_EntityContainer` 
and does not override `::draw()` that seems to answer the question.

I think `RS_Text::update()` would be the place to start looking.
It looks like `RS_Font` and `RS_Insert` are used.
And it looks like `RS_Font` uses `RS_Polyline`.

In the latest code you can disable the scrollbars in application preferences.
This seems to help performance.

2.1.0 is released but not available as an openSUSE package yet.

If you are willing to build from source, I just added an openSUSE section:
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

r-a-v-a-s, 2016-06-07:
It looks like polylines are just arcs and lines;
therefore, someone would need to look at optimizing 
`RS_Line::draw` and/or `RS_Arc::draw`.

jirislaby, 2016-06-07:
I pushed an update to 2.1 in openSUSE. It is pending now.

I installed it too.

I switched off scrollbars in the options->app preferences menu, but it has no effect. Should scrollbars dismiss?

Whatever patches you might have for testing purposes, I will try. No problem to rebuild lcad here.

r-a-v-a-s, 2016-06-07:
The scrollbars won't be removed for any open drawings.
You need to close and reopen the drawing.

jirislaby, 2016-06-07:
I see. So I tested it and it helps a little. No miracles :).

r-a-v-a-s, 2016-06-08:
I've made it so no text is drawn in draft mode 
(instead of based on the conditions I removed). 
We will see if anyone complains about that...
This increases performance a lot for your drawing.

I've also added a hotkey for toggling draft mode: `Ctrl+D`

jirislaby, 2016-06-09:
Oh man, it is much better in the draft mode now.

Now we can optimize the non-draft mode :). Given Autodesk CAD can handle the drawing flawlessly even with text, I believe we can do so with some effort too. If you need any measurements, profiling or anything, just let me know.

jirislaby, 2016-06-09:
BTW do we happen to render all elements, including those currently out of the current view pane? If I zoom in to a single object, I would not expect the rendering to be slow, but it is...

r-a-v-a-s, 2016-06-09:
I was thinking about that the other day;
and I'm inclined to say that there is no ["viewing frustum culling"](https://en.wikipedia.org/wiki/Hidden_surface_determination#Culling_and_Visible_Surface_Determination).
That's out of my realm of experience, 
and I already have several projects I'm working on for LibreCAD.

If anyone else wants to take a look, drawing can be followed by: 
- `QG_GraphicView::paintEvent`
- `RS_GraphicView::drawLayer2`
- `RS_GraphicView::drawEntity(RS_Painter *painter, RS_Entity* e)`
- `RS_GraphicView::drawEntity(RS_Painter *painter, RS_Entity* e, double& patternOffset)`
- `RS_GraphicView::drawEntityPlain(RS_Painter *painter, RS_Entity* e, double& patternOffset)`
- and then something like `RS_Line::draw`

r-a-v-a-s, 2016-06-16:
Actually on quick review of `RS_Line::draw` and `RS_Arc::draw`
I noticed code for: `//only draw the visible portion of line`
therefore, the fact that zooming in doesn't seem to improve performance is odd.

If I was inspired to work on it, 
I might start by making separate classes for 
construction lines and pattern lines.
While it would only be a small optimization,
it would make things less scary (over 100 lines in RS_Line::draw).

r-a-v-a-s, 2016-06-19:
I put some effort into understanding why zooming in didn't improve performance.
In short, I have fixed the problem, and performance will now improve when you zoom in.

In `RS_Line::draw`
- the viewport rectangle was being created redundantly for every line
--  I moved the creation of the viewport rectangle into `QG_GraphicView::paintEvent`.

- it was checking if both points were in the viewport, and if not then clipped the line
-- I found dead code for testing if the entity was outside of the viewport in `RS_GraphicView::drawEntity`
--- I updated / added to the logic to make it work again; thereby, avoiding the call to `RS_Line::draw`
-- I disabled the line clipping from `RS_Line::draw`... it seems unnecessary (but could use profiling)
--- any visible line is simply drawn, this avoids finding which point is in the viewport 
and 4 intersection checks 

I think we can probably make some more optimizations;
however, text is made of many entities, 
so at some point I think the performance becomes based on
how fast your cpu will process them all on a single thread.

If anyone else wants to take a look,
I'll be working on other things the next few days.

rubdos, 2016-06-20:
Did you guys think of trying a quadtree or kd-tree for storing entities? It could perhaps make things (potentially a lot) faster when zoomed in, because it eliminates a lot of boundary checks.

Other than that, I'm looking at the code now (I ran callgrind/kcachegrind) trying to find some things. I'll let you know if I see something.

rvt, 2016-06-20:
Hye, 

for LibreCAD 3 we have implemented a quad tree to improve performance.
I am not sure how hard/easy that would be to implement that in LibreCAD 2.
That said, we do have to implement also hiding small entities, because when zoomed out you will still have to draw all entities. i think it was mentioned somewhere else that some CAD tools support 'don't draw small entities' like small text.

rubdos, 2016-06-20:
> for LibreCAD 3 we have implemented a quad tree to improve performance.

Great

> I am not sure how hard/easy that would be to implement that in LibreCAD 2.

Depending on when you plan to release LC3 and dispose LC2, you should consider this. I can try to investigate how hard it would be, but I suppose there are people with better codebase knowledge than me who could answer that question

> That said, we do have to implement also hiding small entities, because when zoomed out you will still have to draw all entities. i think it was mentioned somewhere else that some CAD tools support 'don't draw small entities' like small text.

You could fallback on draft mode when text gets smaller than ~10-15 pixels in height, and hide it when smaller than ~3 pixels?

rubdos, 2016-06-20:
Please see #772, I'm trying to find more to optimize now.

r-a-v-a-s, 2016-06-20:
>The EntityContainer's in question are those of RS_Line::draw(), which uses one to calculate some intersections before getting disposed. I assume that that part can be optimized easily, although someone could have a look at the mentioned clear() method too.

I eliminated some of what the container was used for;
therefore, I would be inclined to try to eliminated it completely.

It seems like it's still being used for construction lines...
Do we really need to create 4 lines and do line-line intersections for that?
Would using slope-intercept be more efficient?
http://mathopenref.com/coordequation.html
I don't work with math much.

Regarding `clear()`... it ends with `resetBorders()`
which seems pointless if we are deleting the container.

r-a-v-a-s, 2016-06-21:
Text will be drawn as it's bounding box when it's so small it's unreadable.
This makes it so your drawing has no apparent performance issues on this PC.

r-a-v-a-s, 2016-06-21:
I've made it so text is drawn as the bounding box while panning.
Hopefully performance will be acceptable for any system now... :sunglasses:

jirislaby, 2016-06-21:
Works like a charm!

Big thanks to everybody working on this!

closed 2016-08-17 by r-a-v-a-s