LibreCAD/LibreCAD #730: Can't open DGN-files

Jan-80, 2016-05-26:
AFAIK, I can't open DGN-files.
These are CAD-files from Bentley Microstation.
I thought there was an 'open' specification for DGN-files. Maybe it could be incorporated in LibreCAD?

r-a-v-a-s, 2016-06-04:
Adding support for a file format isn't a small job.
Someone who uses that format will need to become inspired to add it.

pgiffuni, 2016-09-10:
This library should do it:
http://dgnlib.maptools.org/
(fine license, documentation and even example files).

It would certainly be useful for people migrating from Microstation to something else.

Jan-80, 2016-09-10:
The reason I asked, was to have an alternative to Microstation to edit .DGN files.

I found an alternative in Bentley Powerdraft, lower cost, but still not cheap. Yet, for some supervisors that insist on editing capabilities, I still have to install a $560/year license of Powerdraft, for occasional use! LibreCAD looked fine, but could not read the well documented .DGN files. 

In general, the fact that an alternative program, like LibreCAD would be able to open the files from the 2 biggest contenders in the CAD world - i.e. Bentley (Microstation) and Autodesk (AutoCAD) - would lead a great deal to it's consideration as an alternative, an eventually, it's acceptance and use. People that switch software package, have to be able to read - or at least transform - their old files. I think it's in the best interest of LibreCAD that those 2 file formats are readable, if not writable.

r-a-v-a-s, 2016-09-10:
The contributors to this project are volunteers.
LibreCAD is not a company; and the founder of LibreCAD 
completely switched focus to a new project "LibreCAD 3" a while ago.
Therefore, it's only in the best interest of the people that need those file formats.

I doubt any current contributors will be taking this on.

Jan-80, 2016-09-10:
Isn't it the aim of the people that write a program that it is largely distributed and appreciated? I think adding those file formats would do that. I had no intention of offending anyone. I am very well aware of the status of the contributors. But if you don't want to hear any more well-meant suggestions, goodbye.

r-a-v-a-s, 2016-09-10:
In short, what I was saying is: 
- you might get lucky, but I wouldn't hold my breath
- you are encouraged to try in implement it if you think it's valuable
-- consider if you want to put the effort in to a theoretically obsolescent program

closed 2016-09-10 by Jan-80