LibreCAD/LibreCAD #731: create perpendicular line

wykhuh, 2016-05-29:
Hi, I just started using LibreCAD. I would like to create a perpendicular line from a selected point to a selected line. In the screenshot, I want to draw a line from Point 1 that is perpendicular to line  B

[image]

In AutoCAD,  there is a snap to [perpendicular option](http://www.mycadsite.com/tutorials/level_4/autocad-osnaps-4-2.htm).

 Is there a similar function in LibreCAD?  In order to draw the perpendicular line, I had to
1) use `perp` to draw a line perpendicular to line B, place the midpoint of the perpendicular line at point 
2) git rid of the  left half of the perpendicular line at point one by using `trim`
3) extend the right half of the  perpendicular line to line B by using `trim`

r-a-v-a-s, 2016-05-29:
It's a useful feature that we don't have...
http://forum.librecad.org/New-contributors-amp-The-LibreCAD-team-td5713777.html

Your trim solution might be the optimal one...
I've been setting the orthogonal line length to 10000
and then I use the snap intersection mode to create the desired line,
and delete the one created with the orthogonal tool.

wykhuh, 2016-05-29:
I would like to help with the documentation. Searching through Github issues, `rs_commands.cpp`,  and the forum on how to do stuff is not the best use of time. Any suggestions on how I can get started?  I never worked in C, so I won't be able to contribute to the codebase quite just yet.

lordofbikes, 2016-05-29:
Yes, it's very similar in LibreCAD, use the orthogonal line tool for this.

1. With the lines A and B drawn use the Orthogonal line tool from the Tools->Line menu or use command `perp` or `ortho`, the cursor changes to a gray cross.
2. Somewhere on the top the tool specific bar appear where you can set the length for the orthogonal line. Change it to your needs.
3. Click on line B to set it as reference for the orthogonal line. The color of line B changes to visualize this.
4. Use a valid snap mode an click on point 1 to place the line. The orthogonal line is placed with it's center in point 1, orthogonal to line B.
  * You can place more orthogonal lines relative to line B until you click the right button or press ESC.
  * Then you can select a new reference for more orthogonal lines
  * or click right or press ESC again to cancel the orthogonal line tool
5. Now use Trim or Trim two to trim the drawing for your needs.

r-a-v-a-s, 2016-05-29:
If you want to help complete http://wiki.librecad.org/index.php/Commands
that would be awesome...

When I rewrote it I made the table formatting consistent.
I don't think you will have a problem seeing the pattern.

The main page has a 'wiki help' section with links.
http://wiki.librecad.org/

Here is the majority of new additions:
http://forum.librecad.org/updating-rs-commands-td5711646.html#a5711654

Feel free to post in the dev section regarding working on documentation.
http://forum.librecad.org/

r-a-v-a-s, 2016-06-25:
I'm going to assume you decided against doing the work.

fa201, 2020-04-12:
@lordofbikes : feature is done and reference documentation is updated.

sand1024, 2026-09-26:
supported by Line from Point to Line action

closed 2026-09-26 by sand1024