LibreCAD/LibreCAD #732: DBL_EPSILON

perepujal, 2016-06-01:
Hi,
Debian Sid(unstable) tried  to compile with 
g++ (Debian 5.3.1-21) 5.3.1 20160528 and 
g++-6 (Debian 6.1.1-4) 6.1.1 20160519

Both complains about DBL_EPSILON in files
lib/math/rs_math.cpp and lib/engine/rs_ellipse.cpp

A comment from Ries on IRC says it may be obsolete

r-a-v-a-s, 2016-06-01:
It builds on 3 different systems. You will need to be more descriptive than 'complains'.

perepujal, 2016-06-01:
Sorry, there is an extract of the output for rs_ellipse.cpp

lib/engine/rs_ellipse.cpp: In member function ‘bool RS_Ellipse::createFromQuadratic(const LC_Quadratic&)’:
lib/engine/rs_ellipse.cpp:793:20: error: ‘DBL_EPSILON’ was not declared in this scope
  if(determinant>= -DBL_EPSILON) return false;
                    ^~~~~~~~~~~
lib/engine/rs_ellipse.cpp:803:30: error: ‘DBL_EPSILON’ was not declared in this scope
  if(qCentered.constTerm()>= -DBL_EPSILON) return false;

r-a-v-a-s, 2016-06-01:
Looks like you are missing float.h
http://www.cplusplus.com/reference/cfloat/

Do you have Qt and C++?

perepujal, 2016-06-01:
I hope so, I have libqt4-dev and libstdc++-6-dev, which provides /usr/include/c++/6/tr1/float.h

r-a-v-a-s, 2016-06-01:
Does the package support C++11?

If you want to be a beta tester, and not a programmer, then you can use the daily ppa:
https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-daily

r-a-v-a-s, 2016-06-01:
Also, did you see the Qt5 build instructions?
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

closed 2016-06-02 by r-a-v-a-s