LibreCAD/LibreCAD #740: problem c/n install librecad on ubuntu 12.04 LTS my t-42 laptop

contentprovider, 2016-06-06:
Ubunut Software Center refused to install libre cad for lack of dependencies. 
The following packages have unmet dependencies:

librecad: Depends: libc6 (>= 2.4) but 2.15-0ubuntu10.12 is to be installed
          Depends: libgcc1 (>= 1:4.1.1) but 1:4.6.3-1ubuntu5 is to be installed
          Depends: libqt4-help (>= 4:4.5.3) but 4:4.8.1-0ubuntu4.9 is to be installed
          Depends: libqt4-qt3support (>= 4:4.5.3) but 4:4.8.1-0ubuntu4 is to be installed
          Depends: libqtcore4 (>= 4:4.7.0~beta1) but 4:4.8.1-0ubuntu4.9 is to be installed
          Depends: libqtgui4 (>= 4:4.6.1) but 4:4.8.1-0ubuntu4.9 is to be installed
          Depends: libstdc++6 (>= 4.4.0) but 4.6.3-1ubuntu5 is to be installed
          Depends: librecad-data (= 1.0.1+nolibs-2) but 1.0.1+nolibs-2 is to be installed
and the installer refuses to continue..

contentprovider, 2016-06-06:
how to get the dependencies installed. and working so that the installer will continue the install
or whatever to get user access to the librecad

r-a-v-a-s, 2016-06-06:
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source
or
https://launchpad.net/~librecad-dev/+archive/ubuntu/librecad-daily

closed 2016-06-07 by r-a-v-a-s