LibreCAD/LibreCAD #745: Dimensions' snap precision _and_ free placement

Cantar4, 2016-06-07:
While laying down dimensions, while looking for the best location for the figures/arrows,  
would it be possible to auto-disable the choosen snap combi 
then enable it back when nailing the dimension in place ?
This way we would avoid both the wrong targeting that free snap (sometimes) induces,
and the time it takes to flip to free snap before placing a dimension.

In programmer terms this feature description would be
"a mode where free snap is automatically activated after the selection
of a dimension's points, and deactivated after the final placement of the dimension"

r-a-v-a-s, 2016-06-08:
Something you can try is enabling Keycode mode (then restart LC).
Then you can type: sf
to toggle free snap, without pressing enter.

Cantar4, 2016-06-08:
The left hand thumb being devoted to the ctrl and shift keys,
I am working with the right hand on the trackpad permanently,
it is less easy to leave it to type sf than hit and click the free snap icon.

r-a-v-a-s, 2016-06-08:
Ah... I'm curious: what would be your ideal method of toggling it (if not the automatic method)?

Cantar4, 2016-06-08:
the spacebar or nothing :)
as a default preference under <mofify layer of selected entity at layer activation> :
it would read <while under dimensions, spacebar toggles freesnap on/off>

Cantar4, 2016-06-09:
if the spacebar option is not available, ctrl n would be workable.

r-a-v-a-s, 2016-06-09:
I'd like to allow for spacebar to be used for accepting commands like in AutoCAD.
http://forum.librecad.org/Do-you-use-the-cal-command-td5713782.html

Ctrl+N is already used for a new file.

Ctrl+M or Ctrl+Space are available.

It makes me wonder about the most practical use of Ctrl+Space.
It could be another way to trigger a custom menu,
and it wouldn't cancel the current action.

Cantar4, 2016-06-10:
ctrl + spacebar is not bad but why to add another operation when 
" a mode where free snap is automatically activated after the selection
of a dimension's points, and deactivated after the final placement of the dimension " 
perfectly and efficiently solves the problem?
(something neither Autocad nor Qcad offer)
30% of the time spent on industrial design 2D drawings is laying dimensions,
anything which speeds up this operation should receive a high priority mark.

r-a-v-a-s, 2016-06-10:
I was just brainstorming when I suggested that... as I am now.
Ultimately my decisions are tactical... the most gain at the least cost.
The hotkey takes a single line of code, 
the automatic feature requires research and consideration.
If I used dimensions, then it would be a higher priority.

Cantar4, 2016-06-10:
_If I used dimensions, then it would be a higher priority._

happy man already living in a dimensionless environment !
I am in Grenoble France, you in LA  USA
Would you come for the café in the next two minutes ? 
I leave the door open.

r-a-v-a-s, 2016-06-10:
It takes minimally two minutes to warm up my astral projector.
The 0th dimension is highly underrated... ;-]

Honestly I haven't used LibreCAD in months,
and I'm having a hard time staying motivated as a contributor.

Cantar4, 2016-06-14:
knowing that the two signals able to command the freesnap activation do exist already
for the screen bottom instructions:
"specify  dimension line location" would also be the 'Snapfree only' trigger,
"specify first extension line origin" would also be 'return to the 'selected snaps',
why not to use them? Is there a risk of side effects?

r-a-v-a-s, 2016-06-14:
Yes that is what you would use;
however, the bridge between 
the dimension actions 
and the snap actions doesn't exist.

Cantar4, 2016-06-15:
is a quickly made monkey bridge enough
or do you need a geopolymer limestones constructed heavy bridge ?

r-a-v-a-s, 2016-06-15:
> would it be possible to type ssf to make sf stick in place? then the space-bar would toggle the Snap-free icon On and Off without having to re-type s and f each time...

I've made it so that space-bar will toggle free snap when the cmd input is empty.

Cantar4, 2016-06-16:
(2.2.0 alpha 26) right after "specify first extension line origin' shows,
snapping on it disactivates the empty command line  
=> no more space-bar action.

r-a-v-a-s, 2016-06-16:
It works perfectly on my PC and iMac... >_<

Space will activate the command line...
so I don't understand what the problem is.

Note that it doesn't work with Keycode mode.

r-a-v-a-s, 2016-06-16:
It will work for keycode mode in alpha-29.

Cantar4, 2016-06-16:
_Note that it doesn't work with Keycode mode._

Yes, that was it!
 and now with alpha-29 it's a joy to lay dimensions with LC .:-)))
--jp

(btw, do you know that the snappers can (abusively) snap
to the extension lines?  
i.e. a wrong horizontal dim can result from a snap to a prior vertical dim.

Cantar4, 2016-06-17:
After actively using this godsend function for a few hours, a light tuning appears as a necessity.
While sf is active, its background becomes dark grey ; it is logical and would 
confirm the snapfree status of the mouse, to remove the dark backgrounds 
of the desactivated snap modes.
This visible change in the peripheric vision of the draftman would help a lot ;
I never forgot to hit spacebar prior to placing a dimension but I sometimes forgot to reactivate 
the snappers thus making errors on the next measurement.

(pity the automatic bridge is expensive to build...)

r-a-v-a-s, 2016-06-19:
The other snap modes are not disabled when 'free snap' is on;
think of it as "snap only when near snap points".

You can change the grey background to any color you desire.
https://github.com/LibreCAD/LibreCAD/wiki/Style-Sheets

```
QToolButton:checked
{
    background-color: lightskyblue;
}
```

You can save the file with TextEdit.
Go into Preferences and selected 'Plain Text'.
Let me know if there is something you don't understand...

sand1024, 2026-09-26:
the issues is outdated now

closed 2026-09-26 by sand1024