LibreCAD/LibreCAD #747: damage files: polyline width

efa, 2016-06-07:
opening the file:
https://sourceforge.net/p/librecad/bugs/_discuss/thread/296ab497/6d12/attachment/LibreCAD_DXFlinesTickness.dxf.bz2
and re-saving, the file become corrupted as polyline width are removed.
If LC do not support polyline width, it should ignore it leaving untouched the property.
See related bug:
https://sourceforge.net/p/librecad/bugs/520/

r-a-v-a-s, 2016-06-07:
Unfortunately Rallaz disappeared. >_<
We really need people willing to work on DXF issues.

efa, 2016-06-10:
will you stay on github or sourceforge as main source/bug tracking?

lordofbikes, 2016-06-10:
The focus has moved from sourceforge to github for the better integration with source code, commits and contributors.
Both have their pros and cons and both are still watched and accepted by the team. So it's up to ones personal preference or where an account exists, which one to use.
If one has accounts in both communities, github is preferred.

FelixNotTheCat, 2017-01-22:
@lordofbikes @r-a-v-a-s  added property to RS_PolylineData for storing the consistent width of a polyline and added line in RS_FilterDXFRW to transfer it for saving, see commit 9c15cbf. -Did virtually the same alongside with thickness and elevation property as the issue with opening and inadvertently discarding line data is the same, see commit 5522d0b. (pls note i tested all with single 

polyline dxf, the submitted dxf was too heavy for my system under debug.)