LibreCAD/LibreCAD #751: Printing into pdf produces empty page

TorstenMr, 2016-06-08:
When I try to print a drawing into a pdf without starting preview first I get an empty page. File size then is 1.2kB. It happens always. It is reproducible.

What I did, step by step:
1. load a dxf.
2. export to pdf -> empty page
3. print to pdf -> empty page
4. preview
5. print to pdf -> works fine

6. load another dxf
7. print to pdf -> empty page
8. preview
9. print to pdf -> works fine

10. load third dxf
11. preview
12. print to pdf -> works fine

fa201, 2020-04-12:
@lordofbikes : this issue should be closed.
I checked on a recent build and it does not happen anymore.

iqqmuT, 2024-08-16:
The issue is still valid with this version:

Version: 2.2.0-undef
Compiler: GNU GCC 13.2.0
Compiled on: Apr  8 2024
Qt Version: 5.15.13
Boost Version: 1.83.0
System: Ubuntu 24.04 LTS

dxli, 2026-09-13:
Fixed in 2.2.1 by ed79434d5: **Print...** and **Export as PDF** now open Print Preview first when it isn't open, and the preview fits and centers the drawing on the page when it opens, so a drawing away from the page area no longer prints as an empty page. The reports above are on 2.2.0 and earlier, which do not have this change.

With the fixed print scale option set, the preview keeps the drawing's saved scale and placement instead; what the preview shows is what gets printed, and **Fit** there puts the drawing on the page.

If you still get an empty PDF on 2.2.1.5 or a current master build, please reopen with the drawing and the steps.

closed 2026-09-13 by dxli