LibreCAD/LibreCAD #755: Changing attributes of a block does not change attributes of sub-entities

Equoda, 2016-06-09:
Attributes of sub-entities doesn't change when apply ```Attributes``` tool to the block of entities.

Steps to reproduce:

1. Create new draft
2. Create a shape (rectangle)
3. Create a shape (rectangle)
4. Create a block with 2. and 3. (select shapes -> ```Create Block -> b1```)
5. Change ```b1``` attributes with ```Attributes``` tool (select ```b1``` -> ```Attributes -> Color: red```)

Result: no visible changes in color.

```Pen``` property has been changed only for ```Insert``` entity corresponding to ```b1```, not for its sub-entities (```RS_Modification::changeAttributes```).

By the way, I can see such pieces of code for sub-entities cloning procedure as:

```.cpp

entity->setPen(RS_Pen(RS2::FlagInvalid));
entity->setLayer(nullptr);

```

Is there were intention to show a block of entities for user using attributes of the whole block/insert, not individual sub-entities? Why such kind of initialization has been used?

Equoda, 2016-06-13:
It's all about:

Version: 2.2.0-alpha-19-gd5b8b19
Compiler: GNU GCC 4.9.2
Compiled on: Jun 13 2016
Qt Version: 5.3.2
Boost Version: 1.55.0

On:  Debian 8 x86_64 GNU/Linux

Equoda, 2016-06-23:
Fixed by pull request #762

kfoltman, 2016-09-02:
I'm not sure if this approach is correct. I mean, the new code as written will change the block *definition*, not just a single instance. Example:

- create a block
- insert the block 3 times
- change the colour on one of the 3 inserted blocks
- save, close and open the project
- now all the instances of the block have the changed colour

I came across it while working on some polyline undo bugs, and I can try figuring it out if you want.

kfoltman, 2016-09-02:
My gut feeling is that the block rendering should be checking if the insert object it belongs to has a pen override and either use the original attributes in the block definition, or the override value specified in the insert object (if present). This way the attribute change in the insert will cause the specific instance of the block to be drawn with the changed attributes, while preserving the block definition itself.

lordofbikes, 2016-09-02:
I agree, that the steps you describe are not correct.
But I have to put in a veto on your approach.

There are no attribute overrides for blocks in DXF reference. A block is a container of entities kept in a separate table of the DXF file. The instance in the drawing is a INSERT entity which has attributes like coordinates of the reference point, the rotation angle and X/Y scale factors. But no attributes like color or layer.
Thus there is no compatible way to store manipulated instances of blocks.
The correct way is to add a modified block to the block table and insert an instance of it.

Concerning the status quo, LibreCAD should apply the modification of a block instance to all other instances immediately, without reopening the drawing. 
Or we should disable the modification of block instances at all and only allow the modification by the block editor. That would make this more clear.

kfoltman, 2016-09-02:
OK - I think I accidentally implemented the 'apply to all instances' approach yesterday, I just need to make sure it works. :) Or I could implement the blocking approach, with some warning perhaps.

But, what's wrong with the approach of automatically cloning/replacing the block, apart from the fact that it's not trivial to implement?

lordofbikes, 2016-09-02:
Well, that sounds different from a pen override approach and this could be implemented without any DXF storing issues.

On attribute modification of 'block' instance ask the user if he wants to change all instances, then apply it to the 'block' reference and update all instances. If not, create a 'copy of block', add it to the block list and replace 'block' instance with a 'copy of block' instance.
We should give the user a chance to rename 'copy of block' when cloning it.

This should work, is compatible in sense of DXF storing and sound good to me.

Equoda, 2016-09-02:
There are duplicated "Pen" properties for "Block" and "Subentities". Should it be? May be better to ignore blocks pen and show entities according to their "native" pen? So, blocks would be only geometry/construction unions without the ability to influence on a visual representation of drawn objects (more like "Inserts").

I mean:
1. Draw entities and color them as you like
2. Make a block with entities from 1.
3. Insert this block to draft a few times
4. Color one of the inserted blocks differently so that all its subentities get one color (technically, change pen properties of subentities ignoring block pen)
5. Have a draft with the originally colored blocks and the modified one

kfoltman, 2016-09-06:
@lordofbikes Another question: what should happen when changing attributes of multiple instances of the same block, either as a single operation or as multiple separate operations. The simplest solution is to make each instance a separate clone. The more complex one would be to make all new instances within a single operation share the same block clone. A crazy solution (and possibly one's that too difficult to implement in reasonable time) would be to find the block "just like this one but with new attributes" and use that instead of creating too many identical clones.

What do you think?

r-a-v-a-s, 2016-09-06:
I don't know about the DXF definition of a block,
but it seems like the value of the idea is that
you have one definition that governs all instances.
If you edit the definition, then all instances update to match.

In that case it seems like it might make sense to limit the modification to:
"Edit the active block in a separate window"

The idea proposed by lordofbikes seems viable.
A button to make a renamed copy of the active block could also be helpful.

It could also be useful to implement the ability to explode blocks to a new or existing layer.
I couldn't find anything about DXF defining sub-layers, which could be useful for grouping.

lordofbikes, 2016-09-07:
@Equoda without knowing the the associated source, I think the pen property of blocks is or should be ignored. It may be there because of inheritance or is an  obsolete artefact from early days.
A block pen doesn't make sense, as a block can contain many entities, different in color, line width and style.
Your steps 1-5 describe exactly what we have discussed before. The problem is to save the modifications. Therefore we must create a modified clone of the block for the block table or explode all the modified instances and save them as independent entities.

@kfoltman it's hard to say, what's the best solution. I played a little bit with blocks and I think the only problem is, that block attributes modified by modify tool doesn't apply to all instances on an open drawing. Editing a block in the block editor does this. When a block is modified in the block editor, all instances are updated on closing the editor.

@r-a-v-a-s I share your understanding of blocks. 

One intention is grouping of multiple entities for easier handling. But the main intention I think is the ability to have multiple block instances controlled by one master. For modification of the master block we already have the block editor, available in the block list.
For block variation we already have the explode tool which split a block into it's basic entities. After modification these entities could be joined into a block again.
Another way is to save a block into a separate DXF file and import it again as a new block to modify.
With an automatic clone function on modification we can simplify this. I think we should ask the user about his intention if he wants to apply modified attributes to the selected entities only (then clone blocks) or if he wants to apply it to all instances (then modify the block master and update all instances).

Forget about my suggestion about limiting block modification. Of course it doesn't make much sense to select multiple extensive block instances from different master blocks and apply e.g. a single color or layer. But I can imagine situations, where a bunch of simple blocks may be changed this way. Having only the block editor can make this a torture.

We already have a cloning method, when using the library browser. Adding a block from library browser creates a new entry in the block list for each instance. To use an imported block for multiple instances, it must be added by the block list. There are pros and cons, but it offers all options to the user.

The main problem I think is to understand the intention behind blocks in sense of CAD. It's different from other software like DTP or pixel/vector graphics in common.

My conclusion is, that we should at least update block instances on attribute modification, this is a bug. 
Having an automatic cloning method with attribute modification may make use more convenient, but it's already possible in the current state. 
If there are unwanted effects, I think it's a matter of detailed work flow documentation or CAD block basics description.

kfoltman, 2016-09-08:
I was looking into it earlier this week, as I wanted to finish my undo bugfixes/refactoring, and changeAttributes was one of the bits affected. There are some difficulties around dealing with blocks, undo and changeAttributes - I think some of the text entities use blocks internally, too, one per glyph from what I can tell. I still don't know some things - like, for example, how undo works when you have a block opened in an editor. Both separate stacks and common stack have potential for bugs and user confusion.

So, I'll keep looking at it, but for the time being I'd go for an easy solution, implementation-wise, just to have at least the undo fixes merged on their own. Then I can try to improve changeAttributes regarding blocks.

closed 2019-05-12 by lordofbikes