LibreCAD/LibreCAD #758: Non sticky construction layer

Cantar4, 2016-06-12:
construction layer activation disappears at file closure

lordofbikes, 2016-06-12:
OS, LC version, steps to reproduce?

As construction layer is a LibreCAD only feature, it's not supported by other CAD and may disappear by opening and saving in an other CAD software.

Cantar4, 2016-06-12:
MacOS Yosemite ; LC 2.1.0 (all beta to the latest stable release) ; between LC and itself ;)
- open 'layer sattings' of the soon to be construction devoted layer ; 
- click 'construction layer', 
- make it red, ok.

This works during the session only, you  must reactivate at file re-opening

lordofbikes, 2016-06-12:
Confirmed.

As I'm sure it worked on 2.0.x, I'll bisect this later, be about to go.

r-a-v-a-s, 2016-06-12:
I don't think it's a bug;
I can't find any code for saving the state,
and it functions the same in 2.0.7.

r-a-v-a-s, 2016-06-14:
Something I just noticed:

```
RS_DEBUG->print(RS_Debug::D_WARNING, "RS_FilterDXF::writeLayers: layer %s saved as construction layer", lay.name.c_str());
```

[RS_FilterDXFRW::writeLayers](http://wiki.librecad.org/lc2doc/class_r_s___filter_d_x_f_r_w.html#a1310817784d455a20bdb96119fcb329f)

r-a-v-a-s, 2016-06-14:
and
```
} else if (isLCdata && (*it)->code() == 1070){
    if ((*it)->content.i == 1){
        layer->setConstruction(true);
    }
```

[RS_FilterDXFRW::addLayer](http://wiki.librecad.org/lc2doc/class_r_s___filter_d_x_f_r_w.html#ad275f9aa0aa43a38931fe942c0eac492)

lordofbikes, 2016-06-14:
Bisecting ended in a couple commits from April last year. As compiling is broken for some commits there, the diff between bad and good is more than 500 KB. It's the libdxfrw update with DWG support, I'll work on it.

lordofbikes, 2016-06-15:
The problem appeared on DXF opening, the construction layer flag was saved correct. Thus the construction layer should be back on old drawings with the fixed version.

r-a-v-a-s, 2016-06-15:
What are those numbers associated with, and how did you know what to change them to?

I definitely wouldn't have found this... kudos :sunglasses:

lordofbikes, 2016-06-16:
The numbers represents the `libdxfrw` version, it's on top of every DXF file written by `libdxfrw`. When reading the file the variables `libVersion` and `libRelease` contain numbers representing the version string. It's like compiler version with major, minor and patch version, but bad implementation.

`libVersion` contains the major version and `libRelease` combines minor and patch version. The implementation concatenated minor and patch as string and then get the number with `toInt()`. Thus `libRelease` was 513 for 0.5.13 when we re-factored the construction layer. Changing `libdxfrw` to version 0.6.0 `libRelease` became 60 which should be more than 513, so <> comparisons failed depending on the generating `libdxfrw` version.
Now I multiply minor version with 100 which is safe up to 0.6.99 :wink:

I think I will split `libRelease` and add macros to build a single version number like compilers or Qt do. That will be clearer and easier to read/understand. Maybe the macros or even the version numbers should be provided by `libdxfrw`, I'll check this.

Cantar4, 2016-06-17:
_Thus the construction layer should be back on old drawings with the fixed version._
Yesss, it works on alpha34 !

lordofbikes, 2016-06-17:
Fine, thanks for catching it.

closed 2016-06-17 by lordofbikes