LibreCAD/LibreCAD #759: Right mouse button click does not select block in Block List

Equoda, 2016-06-13:
No block is set active in ```RS_BlockList``` variable if click right mouse button on a block in the "Block List" widget

Steps to reproduce:

1. Create a new draft
2. Create a shape ("Rectangle")
3. Create a block on 2. (select, "Create block -> b1")
4. Click **right** mouse button on "b1" in "Block List", select "Edit block"

Result: no edit block window opens.

In function ```QC_DialogFactory::requestEditBlockWindow``` ```blist->getActive()``` returns ```NULL```.

If click **left** mouse button on "b1" and then right everything works as expected.

Equoda, 2016-06-13:
It's all about:

Version: 2.2.0-alpha-19-gd5b8b19
Compiler: GNU GCC 4.9.2
Compiled on: Jun 13 2016
Qt Version: 5.3.2
Boost Version: 1.55.0

On:  Debian 8 x86_64 GNU/Linux

matthijskooijman, 2021-02-24:
Looks like this is fixed by #773 which is merged? Should this issue be closed?

lordofbikes, 2021-02-24:
The issue still exists for me in latest code, on Windows.
Either it's OS dependent or the issue found its way back somehow.
Maybe it came back with the multiple selection feature, needs to test code before the merge.
In fact, the block editor opens with the context menu, but not the block which was right clicked.
Instead the currently selected block is opened, which is not correct.

matthijskooijman, 2021-02-24:
Ah, indeed, I can reproduce with current master on Linux too.