LibreCAD/LibreCAD #774: copied objects are not hidden with other layer objects

r-a-v-a-s, 2016-06-24:
1. create a circle
2. copy and paste
3. toggle layer visibility

The original circle is hidden, but the copy is not.

r-a-v-a-s, 2016-06-24:
It works for Move / Copy,
but not for Ctrl+C / Ctrl+V.

Equoda, 2016-06-24:
OK, there is a question.

While catching this bug some weakly related changes in code have been made, like: ```for (...;...;...)``` -> ```for (auto ... : ...)``` or additional ```if (!ptr)``` checks. Nothing critical, I think. Should it be a part of different pull request?

r-a-v-a-s, 2016-06-24:
You can include those changes.

Cantar4, 2016-06-26:
it seems that v2.2.1 still suffers from the same problem :
Ctrl+C/V now correctly transfer propperties but objects pasted that way are not hidden 
with the layer like the "move-copied' or the 'layer to layer' transfered ones are.

Equoda, 2016-06-26:
Pull request #777 merged

r-a-v-a-s, 2016-06-26:
@Cantar4 
There were no changes associated with this in 2.1.1
A full change log was included https://github.com/LibreCAD/LibreCAD/releases/tag/2.1.1

closed 2016-07-30 by r-a-v-a-s