LibreCAD/LibreCAD #776: A newly created block layer is set to current layer, but not its sub-entities layers

Equoda, 2016-06-24:
A newly created block layer is set to current layer, but this is not true for its sub-entities.

Steps to reproduce:

1. Create a new draft
2. Create a new layer and change its color (for better representation), select the layer
3. Create a shape (rectangle)
4. Repeat 2. and 3. repeatedly. You should have some kind of mosaic draft
5. Select any layer (2.)
6. Select all shapes and create a block

Result: created block is on layer 2. and could be hidden by setting layer 2. "invisible". But individual entities of this block are on their initial layers and obey rules of these layers. There is some kind of mess.

I think, this report should not be labeled as a "bug", but better solution should be implemented. May be, creating a new layer for each new block?..

shawncurry, 2019-05-19:
I'm pretty sure this behavior is actually working as designed, and is indeed consistent with other cad systems.   Both the entities and the blocks themselves have a layer property.  You should change the layer of your entities first.

fa201, 2019-05-19:
I reproduced it and I do not see the issue : LibreCAD does what is intended by the user. This behavior is flexible : if you want all rectangles in the same layer you can create only one layer and if you want a specific layer of specific features of the bloc then you get it.

I disagree with Equoda's proposal because if the user **intends** to have this mosaic for a particular reason then LibreCAD will **prevent** this behavior. I am pretty sure this will end up as a bug report if such feature is implemented...