LibreCAD/LibreCAD #779: easy way to create relative orthogonal points

r-a-v-a-s, 2016-06-25:
If you want to create a point 100 units directly above the relative zero,
then you input `@0,100`

It was suggested that we simplify these cases by prefixing with u,d,l,r
e.g. `u100`

However after considering an English independent method,
I decided to implement it as Ctrl + the obvious keyboard arrow.
e.g. input `100` then press [Ctrl] + [Up-arrow].

Additionally, after you have input a distance (100)
the command line will remember that distance;
so you can just hold Ctrl (Command for Mac)
and continue pressing arrow keys.
If that has practical value... let me know ;-]

dellus, 2016-06-26:
Yes, it's useful, especially for drawing walls in a floor-plan.

binoyte, 2016-07-18:
I think it is a good idea for a feature but the 'u100' command is a wrong solution. My way is that missing coordinates should be interpreted as zero : 

    @,100

closed 2016-08-17 by r-a-v-a-s