LibreCAD/LibreCAD #782: Multiline paste into command line

alx9r, 2016-06-25:
Suppose I have a rhombus described by the following commands:

````
polyline
0,0
10,100
110,100
100,0
0,0
close
````

I expected to be able to create the rhombus by pasting that multi-line command string into the command line.  However, pasting the lines into the box and hitting <kbd>Enter</kbd> results in "Expression Syntax Error".

I also noticed that pasting a multi-line string into the command line removes line breaks.

1. Is there a way to paste multi-line commands into the command line?
2. Is there a way to delimit multiple commands on a single line on the command line?
3. Is there some other way to import large numbers of coordinates from a text file to create a polyline?

r-a-v-a-s, 2016-06-25:
This is mostly a duplicate of https://github.com/LibreCAD/LibreCAD/issues/689
Coincidentally I was just thinking about implementing it yesterday.

Currently there is the 'read ascii points' plugin.
http://forum.librecad.org/ASCII-file-format-td5707429.html
http://forum.librecad.org/Reading-ASCII-points-td5706664.html

alx9r, 2016-06-25:
Thanks @r-a-v-a-s.

r-a-v-a-s, 2016-07-13:
You can now load multiline input from a file.
The command line also supports the `;` delimiter now.
You can also assign it to a variable now. 😎 

`p1=polyline;0,0;10,100;110,100;100,0;0,0;close`
Then to use it `\p1`.

r-a-v-a-s, 2016-07-13:
Ctrl+V will paste multi-line input now.
The right-click context menu paste will not.

Also note that you can load a variable file at startup now (application preferences->paths).

alx9r, 2016-07-14:
Cool.  That sounds pretty nice.

r-a-v-a-s, 2016-07-14:
We have [Windows nightly builds][1] if you want to test.

[1]:https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

alx9r, 2016-07-14:
I just tested the nightly build.  The multiline <kbd>ctrl</kbd>-<kbd>v</kbd> works on my machine :sunglasses:  :

[image]

Thanks for the help @r-a-v-a-s

closed 2016-07-14 by alx9r

r-a-v-a-s, 2016-09-10:
ctrl+v will no longer trigger this action,
because it interferes with normal pasting.

It's been added to the command line menu as 'Paste Multiple Commands'.

shouryalala, 2018-02-26:
`Suppose I have a rhombus described by the following commands:

polyline
0,0
10,100
110,100
100,0
0,0
close
``Suppose` I have a rhombus described by the following commands:

polyline
0,0
10,100
110,100
100,0
0,0
close

shouryalala, 2018-02-26:
`yiuuigyuhkil`

shouryalala, 2018-02-26:
`hello`