LibreCAD/LibreCAD #784: hidden layers are not loaded completely anymore

jirislaby, 2016-06-26:
Since I switched from 2.1.0 to the latest git, hidden layers are loaded but incompletely, it seems.

I am loading a dxf drawing which has some hidden layers. When I push the eye button to show the layer, I see only _some_ elements from the layer.

When I downgrade to 2.1, it works normally again.

jirislaby, 2016-06-26:
Up-to-date version can show the elements:
downgrade to 2.1 -> unhide -> save -> upgrade to 20160621 -> load

r-a-v-a-s, 2016-06-27:
I still don't have any idea what caused this, but...
I noticed that mtext would show again if I switched tabs.
This led me to look at `QC_ApplicationWindow::slotWindowActivated`
where I saw `updateInserts()`...
so I added that to the visibility toggle, and now the problem seems to be solved.

jirislaby, 2016-06-28:
Yep, fixed. Thanks!

jirislaby, 2016-10-19:
Hmm, not really, it happened again with 93a23f63fb65256b1aab3ada9c4e5f8df615c34c.

I again have to unhide -> save -> close -> reopen -> see the lines.

r-a-v-a-s, 2016-10-19:
The original issue might be associated with the commits in https://github.com/LibreCAD/LibreCAD/issues/774
They were merged 2 days before this issue was opened.

I can't say why my patch appears to have broken, 
because I don't fully understand what the problem is.

r-a-v-a-s, 2016-10-23:
Can you give steps for reproducing this?
It seems to work as expected for me.

jirislaby, 2016-10-31:
Please download:
http://www.fi.muni.cz/~xslaby/sklad/stavebni_upravy.dxf

and turn on layer KÓTY. I have just found out that if I zoom out, they appear. But they should be there even when zoomed in.

If I turn on the layer, save, close, load, they are indeed correctly drawn.

r-a-v-a-s, 2016-10-31:
It seems to be associated with the viewport clipping.
```c++
    // test if the entity is in the viewport
    if (!isPrinting() &&
        e->rtti() != RS2::EntityGraphic &&
        e->rtti() != RS2::EntityLine &&
       (toGuiX(e->getMax().x)<0 || toGuiX(e->getMin().x)>getWidth() ||
        toGuiY(e->getMin().y)<0 || toGuiY(e->getMax().y)>getHeight())) {
        return;
    }
```

Removing that eliminates the problem;
however, newly created dimensions don't seem affected.

Once the entities become visible, if I zoom or pan
they will disappear when the X or Y axes are out of the viewport.
However, this is not a problem if auto-zoom has been used.
Therefore, it seems like the solution could be in auto-zoom,
which could maybe be applied to the layer visibility toggle.

```c++
/**
 * performs autozoom
 *
 * @param axis include axis in zoom
 * @param keepAspectRatio true: keep aspect ratio 1:1
 *                        false: factors in x and y are stretched to the max
 */
#include <iostream>
void RS_GraphicView::zoomAuto(bool axis, bool keepAspectRatio) {

	RS_DEBUG->print("RS_GraphicView::zoomAuto");

	if (container) {
        container->calculateBorders();

		double sx, sy;
		if (axis) {
			auto const dV = container->getMax() - container->getMin();
			sx = std::max(dV.x, 0.);
			sy = std::max(dV.y, 0.);
		} else {
			sx = container->getSize().x;
			sy = container->getSize().y;
		}
//		    std::cout<<" RS_GraphicView::zoomAuto("<<sx<<","<<sy<<")"<<std::endl;
//			std::cout<<" RS_GraphicView::zoomAuto("<<axis<<","<<keepAspectRatio<<")"<<std::endl;

		double fx=1., fy=1.;
		unsigned short fFlags=0;

		if (sx>RS_TOLERANCE) {
			fx = (getWidth()-borderLeft-borderRight) / sx;
		} else {
			fFlags += 1; //invalid x factor
		}

		if (sy>RS_TOLERANCE) {
			fy = (getHeight()-borderTop-borderBottom) / sy;
		} else {
			fFlags += 2; //invalid y factor
		}
		//    std::cout<<"0: fx= "<<fx<<"\tfy="<<fy<<std::endl;

		RS_DEBUG->print("f: %f/%f", fx, fy);

		switch(fFlags){
		case 1:
			fx=fy;
			break;
		case 2:
			fy=fx;
			break;
		case 3:
			return; //do not do anything, invalid factors
		default:
			if (keepAspectRatio) {
				fx = fy = std::min(fx, fy);
			}
			//                break;
		}
		//    std::cout<<"1: fx= "<<fx<<"\tfy="<<fy<<std::endl;

		RS_DEBUG->print("f: %f/%f", fx, fy);
		//exclude invalid factors
		fFlags=0;
		if (fx<RS_TOLERANCE||fx>RS_MAXDOUBLE) {
			fx=1.0;
			fFlags += 1;
		}
		if (fy<RS_TOLERANCE||fy>RS_MAXDOUBLE) {
			fy=1.0;
			fFlags += 2;
		}
		if(fFlags == 3 ) return;
		saveView();
		//        std::cout<<"2: fx= "<<fx<<"\tfy="<<fy<<std::endl;
		setFactorX(fx);
		setFactorY(fy);

		RS_DEBUG->print("f: %f/%f", fx, fy);

		//        RS_DEBUG->print("adjustZoomControls");
		adjustZoomControls();
		//        RS_DEBUG->print("centerOffsetX");
		centerOffsetX();
		//        RS_DEBUG->print("centerOffsetY");
		centerOffsetY();
		//        RS_DEBUG->print("adjustOffsetControls");
		adjustOffsetControls();
		//        RS_DEBUG->print("updateGrid");
		//    updateGrid();

		redraw();
	}
	RS_DEBUG->print("RS_GraphicView::zoomAuto OK");
}
```

r-a-v-a-s, 2016-10-31:
Adding `container->calculateBorders();` to `RS_ActionLayersToggleView::trigger()` seems to eliminated the problem. I'll push a patch soon.

jirislaby, 2016-11-05:
Nice, fixed.

closed 2016-11-05 by jirislaby