LibreCAD/LibreCAD #786: Crash caused by mouse right click in plugin

Xia-Weiwen, 2016-06-29:
I used doc->getPoint(...) function in my plugin. If I click the right mouse button when selecting a point (actually I want to cancel selection--it is the same action as not in plugins.), the librecad will crash.
Here are the details:
**Env:** Ubuntu 16.04 LTS 64 bit, Qt 5.3.2 (also tried ubuntu 14.04 64 bit and Qt 5.6)
**branch** 2.1 and master
my codes:
+ myplugin.h
```c++
#ifndef MYPLUGIN_H
#define MYPLUGIN_H

#include "qc_plugininterface.h"
#include <QMessageBox>
#include <QPointF>
#include <QPushButton>
#include <QString>

class My_Plugin : public QObject, QC_PluginInterface
{
    Q_OBJECT
     Q_INTERFACES(QC_PluginInterface)
#if QT_VERSION >= 0x050000
     Q_PLUGIN_METADATA(IID "org.librecad.sample" FILE  "sample.json")
#endif

 public:
    virtual PluginCapabilities getCapabilities() const;
    virtual QString name() const;
    virtual void execComm(Document_Interface *doc,
                                       QWidget *parent, QString cmd);

};

#endif // MYPLUGIN_H
```
+ myplugin.cpp
```c++
#include "document_interface.h"
#include "myplugin.h"

QString My_Plugin::name() const
 {
     return ("my plugin");
 }

PluginCapabilities My_Plugin::getCapabilities() const
{
    PluginCapabilities pluginCapabilities;
    pluginCapabilities.menuEntryPoints
            << PluginMenuLocation("my_plugin", "select point");
    return pluginCapabilities;
}

void My_Plugin::execComm(Document_Interface *doc,
                             QWidget *parent, QString cmd)
{
    Q_UNUSED(doc);
    Q_UNUSED(cmd);
    QPointF point;
    doc->getPoint(&point, "Please select a point or input the coordinates");
    // crash here if I click the right mouse button (not the left)
   // do something to the point
}

#if QT_VERSION < 0x050000
Q_EXPORT_PLUGIN2(my_plugin, My_Plugin);
#endif
```
same problem in the plugin example named "Align"
crash happens here:
qc_actiongetpoint.cpp: line 142
caused by null pointer I guess

thank you !

r-a-v-a-s, 2016-06-29:
I eliminated several crashes and other problems in this area. 
https://github.com/LibreCAD/LibreCAD/issues/596

This part of the system really needs more cleaning...

I am unable to produce a crash with the Align plugin now.
Can you give exact steps for reproducing?

Xia-Weiwen, 2016-07-01:
In the Align plugin:
Draw some entities first, if no DXF file is opened.
Secondly we choose some entities (mouse left click on each entity) and press enter. In this step, right click or ESC does not cause crash.
Then, a base point should be selected. In this step, right click or ESC will cause crash.

The problem seems to be in the following files
```doc_plugin_interface.cpp, line1237```
```qc_actiongetpoint.cpp, line 142```
With right click or ESC, no point is selected actually. But the loop in function
```c++
 bool Doc_plugin_interface::getPoint(QPointF *point, const QString& mesage,
                                                           QPointF *base)
```
(doc_plugin_interface.cpp) is over and then the function
```c++
void QC_ActionGetPoint::getPoint(QPointF *point)
```
(qc_actiongetpoint.cpp) still tries to get the coordinate values, from a null pointer I guess.
My plugin causes crash in the same way.

r-a-v-a-s, 2016-07-01:
I added a quick fix. If you see another solution let me know.

Xia-Weiwen, 2016-07-01:
Thanks for the quick fix. I've tried it.
But here is the new problem:
Right click or ESC give a point at the origin (0,0). They do not cancel the selection.

In file ```document_interface.h```, the function ```getpoint()```(line 445) is supposed return false if user cancel the selection (as the comment says). And I thought cancel means right click or ESC. So in my view, right click or ESC should let doc->getpoint(...) return false, and no point is selected.

Maybe something could be done in the loop in function ```getPoint(...)```(doc_plugin_interface.cpp).

r-a-v-a-s, 2016-07-01:
I've added another patch. 

Something to note is that it seems like 
Esc triggers `QC_ActionGetPoint::mouseReleaseEvent`
I can't explain that currently.

r-a-v-a-s, 2016-07-01:
Actually it looks like it's because of
```
        case Qt::Key_Escape:
            slotBack();
```
in `QC_ApplicationWindow::keyPressEvent`

Xia-Weiwen, 2016-07-01:
The code
```c++
if (a->isCompleted() && !a->wasCanceled() )
```
seems not working.
Interesting is that I used debug mode and found that after right click or Esc the value of ```bool canceled``` is 208 (0xD0), not TRUE (which should be 1). And ```!a->wasCanceled()``` returns 209 in this case. Wierd.
So, function ```getPoint``` does not return false in this case.

I've tried many times. I added a qDebug() before the if(), which looks like
```c++
        qDebug("a->wasCanceled(): %u",a->wasCanceled());
        if (a->isCompleted() && !a->wasCanceled())
        {
        ...
        }
```
In most cases, the qDebug() prints 208 (i.e. 0xd0), not 1. In very few cases, ```canceled``` is 1, which is correct.
I use ubuntu 16.04 LTS 64 bit and I copied your newly added codes into librecad-2.1.

Xia-Weiwen, 2016-07-01:
I found something. Qt gives a warning, which says
```
qc_actiongetpoint.cpp:27: In file included from plugins/intern/qc_actiongetpoint.cpp:27:0:
qc_actiongetpoint.h:-1: In constructor 'QC_ActionGetPoint::QC_ActionGetPoint(RS_EntityContainer&, RS_GraphicView&)':
qc_actiongetpoint.h:69: warning: 'QC_ActionGetPoint::canceled' will be initialized after [-Wreorder] 
'bool QC_ActionGetPoint::setTargetPoint' [-Wreorder]
when initialized here [-Wreorder]
 QC_ActionGetPoint::QC_ActionGetPoint(RS_EntityContainer& container,
 ^
```
Move the declaration of ```bool canceled``` before ```std::unique_ptr<Points> pPoints``` and the result seems to be correct.
before:
```c++
private:
    // others
    std::unique_ptr<Points> pPoints;
    bool canceled;
```
after:
```c++
private:
    // others
    bool canceled;
    std::unique_ptr<Points> pPoints;
```
It may have something to do with the  std::unique_ptr.

r-a-v-a-s, 2016-07-01:
It seems to work correctly for me; and there is no reorder warning.
Making the change you suggested produces a reorder warning.

r-a-v-a-s, 2016-07-01:
Note that I'm testing with the Align plugin;
and my previous comment was regarding a build on OS X with Clang.

Now on openSUSE with gcc, I am seeing the 208 returned.

r-a-v-a-s, 2016-07-01:
I confirm that moving the declaration solves that issue... :astonished: 
I can't explain that...

r-a-v-a-s, 2016-07-01:
Another difference: I was using Qt 5.6.1 on the iMac, and 5.6.0 on this PC.

r-a-v-a-s, 2016-07-01:
I appreciate your excellent bug hunting and reporting skills! :sunglasses: 
值得称赞的工作

Xia-Weiwen, 2016-07-04:
Thank you. And thanks for the fix! Could you please push this to branch 2.1 as well?

r-a-v-a-s, 2016-07-04:
Done.

closed 2016-07-05 by Xia-Weiwen