LibreCAD/LibreCAD #787: accurate construction line preview

r-a-v-a-s, 2016-07-05:
It would be useful if a construction line's preview was also an 'infinite' line,
as opposed to only a line segment.

I'm might also make a class for construction lines,
which would optimize drawing.

r-a-v-a-s, 2016-09-10:
I've done so for the '2 points' line tool,
but I've noticed that other tools use RS_Creation....

```c++
void RS_ActionDrawLineParallelThrough::mouseMoveEvent(QMouseEvent* e) {
    RS_DEBUG->print("RS_ActionDrawLineParallelThrough::mouseMoveEvent begin");

    switch (getStatus()) {
    case SetEntity: {
            entity = catchEntity(e, RS2::ResolveAll);
            /*deletePreview();

			RS_Creation creation(preview, nullptr, false);
			creation.createParallelThrough(*coord,
                                           number,
                                           entity);

            drawPreview();*/
        }
        break;

    case SetPos: {
			*coord = snapPoint(e);
            //RS_Vector(graphicView->toGraphX(e->x()),
            //                  graphicView->toGraphY(e->y()));
            deletePreview();

			RS_Creation creation(preview.get(), nullptr, false);
			creation.createParallelThrough(*coord,
                                           number,
                                           entity);

            drawPreview();
        }
        break;

    default:
        break;
    }

    RS_DEBUG->print("RS_ActionDrawLineParallelThrough::mouseMoveEvent end");
}
```

Any ideas on making things more consistent?
Anyone interested in working on this?

fa201, 2023-08-08:
With construction lines, all lines including the circle are not visible.

[image]

When I switch off the construction line, it is OK.
[image]

Version: 2.2.0.2
Compiler: GNU GCC 9.4.0
Compiled on: Jul 29 2023
Qt Version: 5.12.9
Boost Version: 1.75.0
System: Ubuntu 20.04.6 LTS

lordofbikes, 2023-08-09:
@fa201 I just checked Windows 2.2.0.2 and AppImage on Kunbuntu 22.04, both work as expected.
You use the AppImage on Ubuntu 20.04?

OK, confused now, I found #1676. There 2.2.0.2 shows the expected behavior and the issue is addressed to latest `master` branch.

Anyhow, as the issue already has its own place, I close this because it's implemented already.

closed 2023-08-09 by lordofbikes