LibreCAD/LibreCAD #791: Suffix Filters of Open File Dialog Should Be Case-Insensitive

Xia-Weiwen, 2016-07-18:
Hi,  LibreCAD team members.
Now the LibreCAD can only open files with lower-case suffixes, such as .dxf.
However, some other CAD softwares will generate file with upper-case suffixes, such as .DXF. In such cases the users have to rename the files to lower-case suffixes. Otherwise LibreCAD cannot open them (not shown in the file dialog).
The relevant codes are in [librecad path]/librecad/src/ui/qg_filedialog.cpp.
(line
```c++
    fDxfrw2007 = tr("Drawing Exchange DXF 2007 %1").arg("(*.dxf)");
    fDxfrw2004 = tr("Drawing Exchange DXF 2004 %1").arg("(*.dxf)");
    fDxfrw2000 = tr("Drawing Exchange DXF 2000 %1").arg("(*.dxf)");
    fDxfrw14 = tr("Drawing Exchange DXF R14 %1").arg("(*.dxf)");
    fDxfrw12 = tr("Drawing Exchange DXF R12 %1").arg("(*.dxf)");
    fDxfrw = tr("Drawing Exchange %1").arg("(*.dxf)");

    fLff = tr("LFF Font %1").arg("(*.lff)");
#ifdef DWGSUPPORT
    fDwg = tr("dwg Drawing %1").arg("(*.dwg)");
#endif
    fCxf = tr("QCad Font %1").arg("(*.cxf)");
    fJww = tr("Jww Drawing %1").arg("(*.jww)");
    fDxf1 = tr("QCad 1.x file %1").arg("(*.dxf)");
    switch(type){
    case BlockFile:
        name=tr("Block", "block file");
        break;
    default:
        name=tr("Drawing", "drawing file");
```

Could you please enable libreCAD (both branch 2.1 and master) to open files with upper-case suffixes ? Thank you !

r-a-v-a-s, 2016-09-07:
It seems to work for me.

Does a system specific file dialog or the Qt file dialog appear for you?

```c++
QG_FileDialog::QG_FileDialog(QWidget* parent, Qt::WindowFlags f, FileType type)
                            :QFileDialog(parent, f)
{
//# check if system are linux+KDE to use QFileDialog instead "native" FileDialog
//# KDE returns the first filter that match the pattern "*.dxf" instead the selected
#ifdef Q_OS_LINUX
    QDialogButtonBox::ButtonLayout layoutPolicy = QDialogButtonBox::ButtonLayout(this->style()->styleHint(QStyle::SH_DialogButtonLayout, 0, this));
    if (layoutPolicy == QDialogButtonBox::KdeLayout)
        setOption ( QFileDialog::DontUseNativeDialog, true );
    else
        setOption ( QFileDialog::DontUseNativeDialog, false );
#else
    setOption ( QFileDialog::DontUseNativeDialog, true );
#endif
// ...
}
```

I've considered adding an option to use the Qt file dialog.

Xia-Weiwen, 2016-09-07:
I'm using Ubuntu 14.04/16.04 with Unity desktop environment or the LXDE. It shows the native file dialog.
I think the solution is simple. Just add uppercase suffixes into the filters and it will work.
Here is an example:
file: librecad/src/ui/qg_filedialog.cpp, line: 90
original:
```C++
fDxfrw = tr("Drawing Exchange %1").arg("(*.dxf)");
```
modified:
```C++
fDxfrw = tr("Drawing Exchange %1").arg("(*.dxf *.DXF)");
```
Then the file dialog will show files with ".DXF" suffixes.
You may add all uppercase suffixes into the filters.
Thank you!

lordofbikes, 2016-09-07:
Using the Qt file dialog may not solve this issue, as there is an open Qt bug report for that.
See:
http://stackoverflow.com/questions/34858220/qt-how-to-set-a-case-insensitive-filter-on-qfiledialog
https://bugreports.qt.io/browse/QTBUG-51712

It is recommended to use mime filter types, as they are case insensitive by definition. Using `"(*.dxf *.DXF)"` is not a final solution, what about *.Dxf or any other upper/lower case combination, whyever the owner names his files this way.

I'd like to do some more investigation on this before using a quick-and-dirty solution. 
I can confirm this behavior for Ubuntu 16.04 too.

r-a-v-a-s, 2016-09-07:
That bug report states that the "native" or system specific file dialog is the problem,
and that the Qt dialog works.

I confirm that the cross-platform Qt dialog works on OSX and OpenSUSE.
The native OSX dialog also works.
Therefore this is a system specific issue.

Xia-Weiwen, 2016-09-08:
To be specific, I'm using Ubuntu 16.04 LTS with Unity and Lubuntu 14.04 LTS with LXDE.
@lordofbikes MimeType makes sense. The mime-type for .dxf files is "application/x-dxf". If I set mime-type filters like "application/x-dxf", however, the file dialog will show (maybe) all files (I didn't test all types).
@r-a-v-a-s I think you are right. ```setOption ( QFileDialog::DontUseNativeDialog, true );``` works in Ubuntu 16.04 with Unity and Lubuntu 14.04 with LXDE. However bookmarks in native dialog cannot be used if the  option DontUseNativeDialog is set true. It seems that the sidebar URLs in the Qt dialog can only be set before compiling. Enabling adding sidebar URLs dynamically may mean much more work.
Maybe there are better ways :smile:

r-a-v-a-s, 2016-09-08:
I haven't tried it but I think the bookmarks can be saved/loaded with QSettings and
http://doc.qt.io/qt-5/qfiledialog.html#setSidebarUrls
http://doc.qt.io/qt-5/qfiledialog.html#sidebarUrls

lordofbikes, 2016-09-10:
Using mime type filters wasn't valuable, so I added an option to application properties in Defaults tab.
The option is set `on` by default for Linux and `off` for others.
Now it's possible to switch native and Qt file dialog dynamically.

And for the bookmark issue we will find a solution too.

[image]

r-a-v-a-s, 2016-09-10:
@Xia-Weiwen Actually it looks like the Qt dialog automatically saves/loads bookmarks.
I guess you were talking about importing native bookmarks anyhow.

Xia-Weiwen, 2016-09-12:
Thank you very much! I will try the new version. @r-a-v-a-s @lordofbikes

fa201, 2020-04-13:
I cannot reproduce it on Xubuntu 18.04 LTS. Is this still an issue  ?

lordofbikes, 2021-01-31:
This is fixed in latest code.

closed 2021-01-31 by lordofbikes