LibreCAD/LibreCAD #793: favorite colors dockwidget

r-a-v-a-s, 2016-07-20:
This dockwidget will allow for users to maintain a list of favorite colors.
This will allow for quickly:
- changing the active pen color
- changing the color of all selected objects
- selecting all objects of a specific color

r-a-v-a-s, 2016-07-23:
This is a first step... currently this widget doesn't communicate with other widgets.
I'll work on a way to select a favorite on layer creation.

Double-click a favorite color to apply it to the active pen.
Right-click a favorite to access other actions.

Cantar4, 2016-07-23:
(alpha 86 at work) to organize my rainbaw, how can I erase and replace colors 
in the custom pot array  ?  
would it be possible to select / open a pot before pouring in (or washing out) a given color?

r-a-v-a-s, 2016-07-24:
You can organize favorites by dragging and dropping (as of alpha-87).

Right-click -> Remove
I'll probably give the right-click actions hotkeys.

Regarding your last question: 
It's an interesting idea.. I'll think about it.

If you really need to add and remove colors that often,
I would probably implement it as the ability to save/load sets of colors.

Cantar4, 2016-07-24:
(alpha 87 on test)
You are now offering a color selection pattern going through two added stages of selection : 
an alphabetic list carrying all available colors and a list of the current project favorites.

Would you raise your alphabetical list (quite a long lasting poetic reading  ;)  to the upper level 
of the palette construction(*), not leave it at the project list constitution level where it's 
a real burden to search for a color by its native english name.

* when I can take the time to carefully choose a color either into a few classic samples, 
or the rainbow/wheel or your alpha list.

PS: bug noticed : colors added to the alpha list and descriptor augmented here,
become black/white when selected in the project's list.

r-a-v-a-s, 2016-07-26:
Custom names are not supported;
that field is editable so that you can 
type in an existing color name
and it will auto complete.

If you think this widget is useful 
then it was fortunate that I had a brief surge of inspiration.

I won't be contributing much until the tangent point detection is improved.
Work on that hopefully starts sometime next month.

If anyone wants to work on an issue I have self assigned,
just let me know...

closed 2016-08-17 by r-a-v-a-s