LibreCAD/LibreCAD #797: Please add an Apply button to the Current Drawing Preferences window

kharaktur, 2016-07-27:
This would really help, especially when configuring dimensions as you could watch the effect changes had on your drawing and make modifications without having to continually reopen the window.
Thanks for considering this feature.

FelixNotTheCat, 2017-01-25:
@r-a-v-a-s @lordofbikes apply button + functionality added to dwg options dialog, pls see todays commit 94c38f0. feedback/criticism/comments etc welcome and appreciated, thanks

sand1024, 2026-09-28:
close after PR merge, it's about preview

Questions2000, 2026-10-01:
In the past I had thought about mentioning this also for `Application Preferences` window but never did.
If this does get added developers would have to make the decision if this should also be added to other windows such as....
`Application Preferences`
`Widget Options`
etc.
for consistency?

sand1024, 2026-10-01:
@Questions2000 
all these dialogs will go away and be fully replaced.

Questions2000, 2026-10-15:
@sand1024
Okay, you did not say this to the original poster?
Should this post be closed?

You have mentioned a few times that big changes are happening to dialog windows.  Like when I suggesting adding `Widget Options` content into `Application Preferences`.
When you say
>all these dialogs will go away and be fully replaced.
-Are you saying all are going to be combined into 1 dialog?
-Or do you just mean that each dialog is going to replaced but still separate.

To me it seems like the following dialogs would still exist...
-`Application Preferences` + `Widget Options` (combined)
-`Keyboard Shortcuts` dialog
-`Device Options` (I do not use this personally, so not sure)
-`Current Drawing Preferences` (because these settings are separate from application settings)

Will these dialogs above still exist?

Thank you