LibreCAD/LibreCAD #801: LWDISPLAY (System Variable)

superbonaci, 2016-08-09:
Hi, would be possible to display real lineweights? I've got a drawing whose lines are too much thin for my taste. I've increased the line width in the layer but it still looks the same, so I guess the issue is caused by LWDISPLAY

http://help.autodesk.com/view/ACD/2016/ENU/?guid=GUID-18CA6E8F-0008-402E-9F6B-2F9ED8A0A7DD.

lordofbikes, 2016-08-09:
This might be a draft mode issue, see:
http://forum.librecad.org/only-empty-boxes-frames-are-shown-for-image-text-no-line-thickness-td5706737.html
http://wiki.librecad.org/index.php/FAQ
http://wiki.librecad.org/index.php/LibreCAD_users_Manual#DRAFT_VIEW

If you already know about draft mode and are sure it is off, it would be useful if you can share the drawing and maybe a pdf to show how it should look like.

superbonaci, 2016-08-11:
LWDISPLAY
Unknown command: LWDISPLAY

lordofbikes, 2016-08-11:
I think we don't use LWDISPLAY in LibreCAD and *Unknown command* is correct, as the command line doesn't accept variables at all.

Can you attach the DXF file or create a test file which shows your issue?
What does the *List entities* Plugin say about the thin lines? What kind of entity are they?

superbonaci, 2016-08-12:
I see, so LibreCAD doesn0t necessarily support Autocad commands. So, how can I configure LibreCAD to display lines its real width? What would be LbreCAD's counterpart command or option to achieve that, or it's not possible?

closed 2016-08-16 by lordofbikes