LibreCAD/LibreCAD #802: Can't unlock "force Horizontal / Vertical' snap

dgoneill, 2016-08-11:
I have FreeCAD installed on an HP EliteBook with Windows 7 (64 bit). At time, the options to force a horizontal or vertical snap lock become engaged. I do not select them, sometimes they are activated when I first open the program. I cannot deactivate the H/V snap option. When this occurs, there is a command scrolling in the command window saying, "accepted keycode: rr", this message keeps repeating and I can't click the snap buttons to toggle off any of the 3 snap options; "restrict horizontal", "restrict vertical", or "restrict orthogonal".
This seems to be a frequent, random event. Sometimes I can get it out of that mode by closing and restarting libreCAD, other times in starts up in the same mode with the same message scrolling along in the command window.
Is this a 'bug'? Has anyone else had this problem? And, most importantly, does anyone know how I can fix this problem.

Many thanks, dgoneill

r-a-v-a-s, 2016-08-11:
Disable Keycode mode in Application preferences, and then restart.

I can't reproduce the problem.

dgoneill, 2016-08-11:
In Application Preferences, keycode mode is NOT selected in the startup options on the Default tab.Is there another place where I can disable this?
 Dave

r-a-v-a-s, 2016-08-11:
Nope. 

Please copy the information from Help->About

r-a-v-a-s, 2016-08-11:
If you type `rr` is it automatically accepted?

dgoneill, 2016-08-11:
Copied from Help - About:
Version: 2.1.1Compiler: GNU GCC 4.9.1
Compiled on: Jun 25 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 7
Below, I copied the window, you can see the buttons activating the various orthogonal snap restrictions selected. Clicking them merely causes them to blink off and then right back on. Also, the command window scrolls for a bit repeating the same text, "Accepted keycode: rr", as I previously mentioned. Sometimes when I open LibreCAD, this does not occur and I can try the program out for a short while. Eventually, the problem always seems to appear.
Thanks again, Dave

 Dave

r-a-v-a-s, 2016-08-11:
Please use the github interface, as opposed to email.
The attachment didn't come through.

This is very bizarre behavior.
Keycode mode means commands of 2 characters are automatically accepted.

If Keycode mode is not enabled,
then "Accepted keycode: rr" would in theory never be seen.
Ignoring that, the only thing I can think of is your `r` key is malfunctioning,
or some programmable button or script is sending `r`.

Things I would try:
- try a different keyboard
- enable keycode mode (and restart)
- clear settings

r-a-v-a-s, 2016-08-11:
You could also try the latest alpha version:
https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

dgoneill, 2016-08-11:
r-a-v-a-s,

I tried toggling the command option on startup w/ restarting the program. No change. I am running a laptop that is docked with a full size keyboard. I disconnected the keyboard, no change. Then, I closed the laptop lid and used the USB mouse, no change. BTW, I have had this problem since I first installed LibreCAD a few months ago, needless to say I haven't been able to spend much time learning the program. Actually, I had given up, but decided to try again.

dgoneill

r-a-v-a-s, 2016-08-11:
The fact that no one else has reported this,
and that it seems to be random,
suggests a hardware issue.
The code doesn't randomly generate keystrokes.
Although, I really can't understand how it would be showing 
`Accepted keycode: rr` when Keycode mode is disabled.

r-a-v-a-s, 2016-08-11:
What happens when you enter in the `rn` command?

dgoneill, 2016-08-11:
I entered that command as well as the rr command. I have to read quickly, the response says that the command is accepted, but it doesn't solve the problem. I have to read quickly because the other message I mentioned earlier, "Accepted keycode: rr" repeatedly appears and the response from 'rn' scrolls off the screen.

r-a-v-a-s, 2016-08-11:
If you disable keycode mode (and restart)
are you able to type into the command line
and have the text remain there?

dgoneill, 2016-08-12:
No, after typing into the command line, the new messages (from somewhere) caused what I typed and the response to scroll away.
Some hardware information. My computer is a laptop that is plugged into a dock which connects it to a full size keyboard, a large screen, wired internet, etc. I checked what happens when I remove the laptop from the dock so that it is on it's own. Then, I ran LibreCAD and the orthogonal snap issue was not a problem - at least not the 1 time that I tried this before I ran out of time. I'll try again over the weekend. This is not a solution since CAD typically requires working on a larger screen. But, hopefully this may give a hint as to the problem and it's solution.
Thanks for you help.
dgoneill

r-a-v-a-s, 2016-08-14:
You can try [alpha-96][1].
I've rewritten the keycode system, for various reasons;
you can find the toggle in the command line's new menu 
(restart is not required now).

I think it was smart to test undocked... that could be the issue.

[1]:https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/LibreCAD-Installer-2.2.0-alpha-96-g7fbfe27.exe/download

r-a-v-a-s, 2016-08-21:
Did anything change?

closed 2016-08-23 by r-a-v-a-s

dgoneill, 2016-08-27:
No, nothing changed. I was out for a 1 week vacation. No access to internet.

r-a-v-a-s, 2016-08-27:
Then I suggest more testing while undocked.

I'm 99% sure it's not a problem with the code.