LibreCAD/LibreCAD #803: Save after Undo does nothing

Zmuogs, 2016-08-13:
Steps to reproduce:
 - draw two entities,
 - save in to file,
 - close LibreCad,
 - start LibreCad again,
 - open saved drawing,
 - delete one of entities,
 - click Ctrl+S (save),
 - click Ctrl+Z (undo),
 - again click Ctrl+S (save),
 - close LibreCad.

If you open that drawing again then you will notice that Undo before Save had no effect: deleted entity is gone.

-----------------------------------------
Version: 2.2.0-alpha
Compiler: GNU GCC 5.3.1
Compiled on: Jun 10 2016
Qt Version: 5.5.1
Boost Version: 1.58.0
System: Ubuntu 16.04.1 LTS

Equoda, 2016-08-18:
This happens even if LibreCAD stay opened during operations.

This doesn't happen if to save to different files:

 * draw two entities,
 * save in to file (file-1.dxf),
 * close LibreCad,
 * start LibreCad again,
 * open saved drawing,
 * delete one of entities,
 * save in to file (file-2.dxf),
 * click Ctrl+Z (undo),
 * save in to file (file-3.dxf),
 * close LibreCad,
 * open file-3.dxf.

So, may be error is in "save" sequence?

r-a-v-a-s, 2016-08-18:
Something else to note is that if you add an entity 
after the undo and before the final save,
all 3 entities will be saved.

closed 2016-08-24 by r-a-v-a-s