LibreCAD/LibreCAD #805: No dimensions and no text using Win 10. Anybody help? I'll try using portableapp'

Walt5440, 2016-08-18:


r-a-v-a-s, 2016-08-18:
Please copy the information from Help -> About.

Please be more specific about the problem.

Walt5440, 2016-08-18:
Version: 2.1.1
Compiler: GNU GCC 4.9.1
Compiled on: Jun 25 2016
Qt Version: 5.4.1
Boost Version: 1.60.0
System: Windows 

I follow detailed video and text tutorials and make a simple horiz line by selecting 2 pts. Highlight Free Snap, Snap on end pts. Select dimension, Horiz,  select end pts and offset. I get a rectangle with no arrows and no dimensions. 

Tried Librecad in portableapp. Same no dimension but now have text.

Win 10 installed July 15,2016, Updated Aug 15,2016.
thx

r-a-v-a-s, 2016-08-18:
Try inputing larger values for:
Options->Current Drawing Preferences->Dimensions
- Text Height
- Arrow Size

Walt5440, 2016-08-19:
Thank you. That worked.

closed 2016-08-19 by Walt5440