LibreCAD/LibreCAD #808: DXF File open fails to display on attached DXF file. Layers and Blocks seem ok in list.

gh-jmg, 2016-08-25:
DXF import fails to render on attached DXF (exported from Mentor PADS)

Layers and Blocks seem to list on the right hand side ok, in various assigned colours, but there is no visible info following load, even with zoom extents.
No error messages pop up.
[SOT23_Stacks_2D_CC_Full.zip](https://github.com/LibreCAD/LibreCAD/files/438217/SOT23_Stacks_2D_CC_Full.zip)

This file does load into A9CAD & QCAD ok (well, 99% ok, in QCAD everything is visible, but parts on the reverse side seem to have PAD shapes render not the same as front side, & Mirror330 test text is placed wrong - minor issues)

r-a-v-a-s, 2016-08-26:
We need people who are interested in enhancing / debugging DXF and DWG support.
We don't have any truly active code contributors at the moment.

https://github.com/LibreCAD/LibreCAD/tree/master/libraries/libdxfrw
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

gh-jmg, 2016-08-26:
OK, to help, I did some more tests, and removed lines until it became visible, and the two boundary cases are attached.

Seems the Display-fix trigger, strangely, is **removing the 999 DXF comment record** 
```
999
 Some comment here 
```
[Text_Edit.zip](https://github.com/LibreCAD/LibreCAD/files/438635/Text_Edit.zip)

and this larger DXF file, has ONLY the 999 removed, to check if any other issues are hidden.

[Text_Remove999.zip](https://github.com/LibreCAD/LibreCAD/files/438649/Text_Remove999.zip)

Seems all that is needed for fix here, is to safely ignore the 999 comment record ?

gh-jmg, 2016-08-26:
More info: I also loaded the file in post #1, then saved, in A9CAD, and that saved DXF has no comment lines, and it then loads & displays ok in LibreCAD.

gh-jmg, 2016-08-27:
A quick glance at the code shows nothing obvious.
The code has this 
`bool dxfReader::readRec(int *codeData) {
...
    else if (code > 998 && code < 1009) //skip not used at the v2012
        readString();
`
In other places 1000..1005 seem slightly different, but if I try any of 999,1000,1005 the result is No-display of text

Looking carefully at the boundary cases, I notice the fail case also drops the layer info, so maybe the effect is to skip the whole section ?

Digging more I find:
`            if (sectionstr == "SECTION") {
..              if (code == 2) {
                    } else if (sectionstr == "ENTITIES") {
                        processEntities(false);
->
bool dxfRW::processEntities(bool isblock) {
...
    if (code == 0) {
            nextentity = reader->getString();
    } else if (!isblock) {
            return false;  //first record in entities is 0
   }
`
Which seems to assume first record is 0, but NOT if that record is a comment as here.
Maybe the easiest fix, is to add a Skip-on-comment upstream of these sections, in the readRec.
Code Comments suggest there was a skip here earlier, and they added some 1000+ support
Perhaps the comment 999 is collateral damage in that change ?

`   bool dxfReader::readRec(int *codeData) {
...
    else if (code == 999)   // new, simply skip over DXF.comment
// skip by readRec() for get next code ?
    else if (code > 999 && code < 1009) //skip not used at the v2012  1000..1005 extended tags
        readString();
`

r-a-v-a-s, 2016-09-06:
Something I noticed when opening your first file:
`RS_FilterDXF::numberToColor: Invalid color number given.`

gh-jmg, 2016-09-06:
Thanks for trying that.

Did you (re) implement a skip ? (that seems suggested by source comments to have been there in an older version, but my feeling is removed in too-many cases)

Given I tested with manual removal of the 999 line pairs, and it then works, it is hard to see how a 999-read-skip in the right place, cannot get the same result ?
Unless there is more than one place needing to skip ? - one test could be to count the comment-skips, and check against the test file ?

r-a-v-a-s, 2016-09-07:
libdxfrw.cpp
```c++
bool dxfRW::processDxf() {
    DRW_DBG("dxfRW::processDxf() start processing dxf\n");
    int code;
    bool more = true;
    std::string sectionstr;
//    section = secUnknown;
    while (reader->readRec(&code)) {
        DRW_DBG(code); DRW_DBG(" processDxf\n");
        if (code == 999) {
            header.addComment(reader->getString());
// ...
}
```

The code seems to be designed to handle these comments;
although, I confirm your text test works with the comment removed.

Possibly we need to understand:
```c++
bool dxfReaderAscii::readString() {
    type = STRING;
    std::getline(*filestr, strData);
    if (!strData.empty() && strData.at(strData.size()-1) == '\r')
        strData.erase(strData.size()-1);
    DRW_DBG(strData); DRW_DBG("\n");
    return (filestr->good());
}
```

gh-jmg, 2016-09-07:
Yes, the parser I think _tries_ to do some things with codes 
(code > 998 && code < 1009)
but in other places it here
processEntities(bool isblock) {
...
    if (code == 0) {
            nextentity = reader->getString();
    } else if (!isblock) {
            return false;  //first record in entities is 0
   }
it exits on any code <>0, so a 999 will fail here.
I stopped checking here, as there may be other places that are added/bonus-999-code intolerant.

Because the original reader 
readRec(int *codeData)
seems to suggest it used a skip (ie never return with a 999, to confuse later sections, instead simply read-onto-next Code), so I was thinking adding back that Skip option, just for 999, would patch things enough to work, and not need too much code work.

It only seems to make sense to try and readString on a comment, if that is going to be either preserved and saved, or used as some pragma ?
I think LibreCad does neither, at the moment.

lordofbikes, 2016-09-08:
Comments in DXF files seems to be unusual, as this wasn't an issue before.

I fixed this by ignoring comments when reading DXF files. When the drawing is modified and saved in LibreCAD the comments will be lost anyway as there is no relation between reading and writing back the DXF. There is nothing like entity order in a DXF file, modifying the drawing may change the entities order and we don't know where to place the comments on writing the modified drawing.

The only comment libdxfrw supports is the first record, where the libdxfrw writer notes its version for compatibility issues. LibreCAD uses this information in `rs_filterdxfrw.cpp`. 
QCAD does or maybe did the same, QCAD generated DXF files, and thus early LibreCAD files, are commented with 'dxflib'.

I could load all the files you posted with this issue. The SOT23 file needs some time to load, that's because of the many blocks and text entities.

I think this issue is solved by my commit and could be closed when you can confirm it works.
If there are any other issues concerning the content of the files please open new issues.

lordofbikes, 2016-09-09:
You can get the Windows installer now from nightly builds:
https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/

gh-jmg, 2016-09-09:
> Windows/NightlyBuilds/master/

Thanks, downloaded that, and yes, it has solved the Fail-to-load file with
999 comment issue.
Info is now visible.