LibreCAD/LibreCAD #809: Librecad freezes opening my DWG file

frankdownunder, 2016-08-28:
I ran it under control of the debugger and got this:

(gdb) r
Starting program: /home/frank/develop/LibreCAD/unix/librecad 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
RS_DEBUG::setLevel(3)
RS_DEBUG: Critical
RS_DEBUG: Errors
RS_DEBUG: Warnings
[New Thread 0x7fffeb98f700 (LWP 3035)]
[New Thread 0x7fffe1653700 (LWP 3036)]
[New Thread 0x7fffe0e52700 (LWP 3037)]
[New Thread 0x7fffdbfff700 (LWP 3038)]
[New Thread 0x7fffd9056700 (LWP 3040)]
[New Thread 0x7fffd8855700 (LWP 3041)]
[New Thread 0x7fffca1bd700 (LWP 3042)]
[New Thread 0x7fffc99bc700 (LWP 3043)]
[New Thread 0x7fffc91bb700 (LWP 3044)]
[New Thread 0x7fffc89ba700 (LWP 3045)]
void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Delete selected" under id 36
void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Fullscreen" under id 38
[Thread 0x7fffd9056700 (LWP 3040) exited]
QG_LayerWidget::update: nullptr lastLayer
[Thread 0x7fffd8855700 (LWP 3041) exited]
[New Thread 0x7fffd8855700 (LWP 3324)]
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
QG_LayerWidget::activateLayer: invalid layer or nullptr layerView
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_EntityContainer::optimizeContours: hatch failed due to a gap
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
RS_Hatch::update: requesting pattern: not found
RS_Hatch::update: requesting pattern: not found
terminate called after throwing an instance of 'std::length_error'
  what():  vector::reserve

Thread 1 "librecad" received signal SIGABRT, Aborted.
0x00007ffff5c87418 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.

Im happy to upload the DWG file somewhere if thats of use.

saroele, 2016-08-31:
I did not activate the debugger, but same 'freeze' issue here. 
I can upload the dwg if that's of use. 

Version = 2.0.8

r-a-v-a-s, 2016-08-31:
We need people who are interested in enhancing / debugging DXF and DWG support.
We don't have any truly active code contributors at the moment.

https://github.com/LibreCAD/LibreCAD/tree/master/libraries/libdxfrw
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

While it might not help with this issue,
upgrading to 2.1.1 is highly recommended.

frankdownunder, 2016-09-06:
I am using 2.2.0, Ive looked at the stack again, and here is the code it fails on.  In the source file
libraries/libdxfrw/src/drw_entities.cpp,  function DRW_Hatch::parseDwg, 
spline->nknots has the value -65085440 which is a strange value unsuitable to call reserve.  I think if a spline is crazy kaputt, it wouldnt hurt to ignore it, how do I do that?
                } else if (typePath == 4){ //spline
                    addSpline();
                    spline->degree = buf->getBitLong();
                    bool isRational = buf->getBit();
                    spline->flags |= (isRational << 2); //rational
                    spline->flags |= (buf->getBit() << 1); //periodic
                    spline->nknots = buf->getBitLong();
                    spline->knotslist.reserve(spline->nknots);        <------

r-a-v-a-s, 2016-09-06:
Ignoring the spline could work, but ideally you would determine why getBitLong() produces garbage values.

martin-frbg, 2016-11-10:
I just saw something that could be related - in my case, with a dwg from aerofred.com (rc plane enthusiasts' plans exchange) that - apparently coming from some non-Autodesk CAD software - probably lacked some entries. As it turns out, getRawChar8() in dwgbuffer.cpp does not zero its buffer before trying to read, so it will return random data when the read fails (in my case leading to random crashes elsewhere in the code as it tried to new/delete based on bogus values). getRawChar8 is used in one of the code paths in getBitLong...

FelixNotTheCat, 2017-01-27:
@frankdownunder for #826 i made a -simple- amendment to the RS_EntityContainer::optimizeContours() function (a3cd7b5) - it's not been discussed with the LC team though. but if you upload the dwg (it helps if its striped of everything else other than the hatch) i can check if it applies to this hatch shape too. thanks

doudz, 2021-02-04:
I have the same problem with 2.2.0-rc2
```
RS_FilterDXFRW::endEntity(): updating hatch failed: invalid hatch area
terminate called after throwing an instance of 'std::length_error'
  what():  vector::reserve
Abandon (core dumped)
```