LibreCAD/LibreCAD #810: Dimension lines and text disappear when zooming in

ARNBender, 2016-09-06:
I recently noticed that the dimension lines and text disappear when I zoom into my drawing. This wasn't always the case, I used to zoom in and out freely, without fear of losing my dimensions. After playing around with the settings in my LibreCAD, I discovered that this was because the default dimensions settings are set to 'By Block' for the colour and width of the: text size and position; extension lines; and dimension lines, arrows, and ticks. Once I set these to 'By Layer' the dimensions and their related text stay visible when I zoom in.

However, these settings have to be updated in 'Current Drawing Preferences' every time I open LibreCAD. I don't know if this is a bug or if there is an easier way to fix the issue, but would appreciate some insight/ fixes.

I have included my version information below:

Version: 2.2.0-alpha-98-g7a4b5bb
Compiler: GNU GCC 6.1.1
Compiled on: Sep  6 2016
Qt Version: 5.7.0
Boost Version: 1.60.0

Thanks!

r-a-v-a-s, 2016-09-07:
I can't reproduce this.

ab-lcad, 2016-10-27:
Getting the same behaviour. Using Modify->Properties on the affected items seems to solve the problem, as does changing the default settings in Current Drawing Preferences as mentioned above. When I open the Properties of affected items they show that they are set to By-Layer already, but clicking OK seems to reset something & the items display OK at all zooms.

Win7 64 Bit
LibreCAD V2.1.3

robzach, 2017-04-23:
Had the same problem ARNBender mentions; it was solved by simply opening the drawing preferences and clicking OK.

Version: 2.1.3
Compiler: Clang 8.0.0
Compiled on: Sep 22 2016
Qt Version: 5.6.1
Boost Version: 1.60.0
System: OS X Yosemite (10.10)

ScumbagMark, 2017-08-07:
One way of triggering this bug is to copy some text.  As stated by ab-lcad, Modify/Properties fixes the problem (for me anyway).

I'm using Win7 64 bit, and my units are set to mm - just in case that's relevant.

amirm, 2017-11-07:
Quick fix - delete the object and undo the delete. works for me.
Click, DEL, CTRL-Z. Can´t be easier.

However... I lost a little trust in LibreCad. Do I have to check for missing objects when I zoom in/out?? 

Librecad 2.1.3 on Win7-64

BeansnRice, 2017-12-05:
Opening preferences and then selecting OK worked initially for me, but then the issue appeared again.

@amirm solution worked for me and is lasting. I share the sentiment regarding checking for missing objects.

Version: 2.1.3
Compiler: GNU GCC 7.1.1
Compiled on: Jun 26 2017
Qt Version: 5.9.3
Boost Version: 1.64.0
System: Arch Linux

JohnCC330, 2017-12-09:
Same issue here (version 2.1.3). Notably:
- It seems to start when parts of the image. In the original, texts show correctly. They disappear in the copy (Quite predictably)
- In my case, opening one of the texts' properties and Ok solves the issue temporarily, also for all the other texts.

ihatetoregister, 2018-03-04:
Same here, but has only seen this on text objects. Using:
Version: 2.1.3
Compiler: Clang 8.0.0
Compiled on: Sep 22 2016
Qt Version: 5.6.1
Boost Version: 1.60.0

OS: macOS 10.13.3
Graphics: Intel HD Graphics 3000 384 MB

doak, 2018-04-19:
I have found a more robust way to work around (redisplay all entities) at least for each single layer. Previously, I had to find a zoom level which displays all entities, open 'Modify attributes' and accept it without any changes:
To redisplay all entities of a specific layer just open properties of that layer and accept it without any changes.

My version (I compiled myself to patch another bug, see https://github.com/LibreCAD/LibreCAD/issues/936#issuecomment-373961485):
Version: debian/2.1.2-1-doak
Compiler: GNU GCC 7.3.0
Compiled on: Mar 18 2018
Qt Version: 5.10.1
Boost Version: 1.62.0
System: Debian GNU/Linux buster/sid

dvtomas, 2018-05-18:
Same problem here, happens to me very often and is quite annoying. @doak thanks for the workaround.

Version: 2.1.3
Compiler: GNU GCC 7.3.0
Compiled on: Mar 24 2018
Qt Version: 5.10.1
Boost Version: 1.62.0
System: Debian GNU/Linux buster/sid

dvtomas, 2018-05-18:
Found a simple way to reproduce. See this screencast:

https://www.youtube.com/watch?v=RzuoQeSQus8

Hope it helps.

vwren, 2018-06-03:
I can reproduce this regardless of zoom (2.2RC1 on Win7-64).  It appears to be unrelated to zoom, but more to the object's clipping window.  For some reason, dragging items appears to cause the clipping window to be smaller than the viewing window (or offset from it in the opposite direction to what you drag).  When you go into layer properties and accept without changes, it resets the clipping window of the object to match the display window.  Dragging the item again causes the same issue all over again.  This is easiest to see with text objects, because as you pan the window, it actually erases one letter at a time (or one line at a time if panning vertically on multi-line text)

In fact, I can control it.  If I drag an item horizontally, the clipping is offset the opposite direction by the same amount, causing what I dragged to be cut off in the middle of the screen.  If I drag it vertically, the same happens.  Again, selecting layer properties and accepting without changes appears to reset the clipping window.  Testing dragging often leads to a crash.

doak, 2018-06-28:
@vwren, thanks a lot for the analysis.
Would be very Jahre nice if some developer cares about that issue...

doak, 2018-07-07:
I am still looking for the most stable workaround -- and I have found another one: Enter and leave 'Print Preview'.

This seems to restore the clipping windows mentioned by @vwren as well and seems to work for all layers and all objects (even if not visible right now due to this issue).

My current version (Debian testing):
    
    Version: 2.1.3
    Compiler: GNU GCC 7.3.0
    Compiled on: Mar 24 2018
    Qt Version: 5.10.1
    Boost Version: 1.62.0
    System: Debian GNU/Linux buster/sid

doak, 2018-07-07:
Some more observations:

> it actually erases one letter at a time (or one line at a time if panning vertically on multi-line text)

@vwren, it don't get the letter-by-letter erase, though line-by-line works. If I drag horizontal, just the whole block disappears.

Furthermore you can see the offset between clipping and viewing windows while dragging some text, because the bounding box of the text is drawn at "the other" location.

eddyverl, 2019-04-01:
You can solve this problem simply by clicking AutoZoom.

doak, 2019-04-01:
> You can solve this problem simply by clicking AutoZoom.

Wow, workarounds are getting better and better ;)

fa201, 2019-04-11:
I have the same issue on Version : 2.2.0-alpha
Compilateur : GNU GCC 7.3.0
Compilé le : Jun 18 2018
Version de Qt : 5.9.5
Version de Boost : 1.65.1
Système: Ubuntu 18.04.2 LTS
Another work around is to mask and show any layer if you do not want to change the zoom factor.

bchociej, 2019-06-12:
Version: 2.1.3
Compiler: GNU GCC 8.2.1
Compiled on: Nov  8 2018
Qt Version: 5.12.3
Boost Version: 1.68.0
System: Arch Linux

I get the problem where text appears only depending on zoom. Entering & leaving print preview is my current workaround.

PR #1181 (2020-05-29): Modifications: calculate borders of new entities
Fixes #810
Fixes #1146
closed 2020-05-29 by lordofbikes