LibreCAD/LibreCAD #811: creating a new block from an existing block instance

r-a-v-a-s, 2016-09-06:
1. create some entities `i` and make a block `a`
2. insert an instance `ii` of `a`
3. create a new block `b` based on `ii`.

Toggling the visibility of `a` toggles `i` and `ii`.
Toggling the visibility of `b` toggles `ii`.

Any new instances of `b` will also be toggled by `a`.

Entities added to `a` will be added to instances of `b`.
Entities added to `b` will *not* be added to instances of `a`.

Is this desirable behavior?

Equoda, 2016-09-07:
Yes, because the second block is created on the first one. So, hiding parent (first) block you hide content of child (second), but not in reverse.

For example:
1. Create some entities
2. Create a block on 1
3. Insert a copy of the block 2
4. Create some entities
5. Create a block on 3 and 4
6. Try to hide block 2 or block 5 and observe the result

If you want to create an independent block: insert a copy, explode and create a new.

r-a-v-a-s, 2016-09-07:
Is this parent-child block relationship part of the DXF definition?

r-a-v-a-s, 2016-09-07:
I think I get it now... block `a` is simply considered 1 entity in the definition of block `b`.

closed 2016-09-07 by r-a-v-a-s