LibreCAD/LibreCAD #814: Crashes on OSX

Avec112, 2016-09-10:
Versjon: 2.1.1
Compiler: Clang 7.3.0
Compiled on: Jun 25 2016
Qt Version: 5.6.1
Boost Version: 1.60.0
System: OS X El Capitan (10.11)

LibreCAD just randomly crash. I happens when I trigger an event. Have not seen a pattern.

r-a-v-a-s, 2016-09-10:
Are you switching between programs?
I noticed that seemingly randomly when switching back to LibreCAD
the current tab won't be activated,
thereby causing a crash when selecting a tool.
I'll add a patch for it, although I can't seem to reproduce it currently.

You can try alpha-99, which includes a fix for a plugin crash.
https://sourceforge.net/projects/librecad/files/OSX/2.2/2.2.0/alpha/

r-a-v-a-s, 2016-09-10:
Please be more specific about "when I trigger an event".
Do you mean activate a tool?
Are you using a standard or custom toolbar?
Are you using commands?

Avec112, 2016-09-10:
If I click with my mouse (trigger an event) the program just quit. It happened 5-6 times or something. 

I did switch between programs (browser and LibreCAD).

I might know what made the problem happen. I started out with norwegian text at first. I later switched to english. After that the problem did not happen again. Could be a coincidence of course.

Avec112, 2016-09-10:
I take it back. 

I just picked the MText tool, and it crashed.

r-a-v-a-s, 2016-09-10:
If it's the first thing you did after switching back from the browser,
that's probably the issue.
Otherwise, try to remember any steps prior to selecting MText.

r-a-v-a-s, 2016-09-11:
Alpha-102 includes a possible fix.

r-a-v-a-s, 2016-09-11:
If there is a .crash file in
/Users/your_user_name/Library/Logs/DiagnosticReports
that might help.

Avec112, 2016-09-11:
[LibreCAD_2016-09-10-105329_ducati.crash.zip](https://github.com/LibreCAD/LibreCAD/files/465807/LibreCAD_2016-09-10-105329_ducati.crash.zip)

r-a-v-a-s, 2016-09-11:
Thanks; It looks like my guess was correct.
Alpha-102 in theory won't crash...
however, the drawing still won't be activated until you click on it.

Are you using tab mode?

Because this seems like a Qt / Window Manager bug,
I'm not sure if QMdiArea::activatePreviousSubWindow() 
will activate the correct subwindow when more than one is open.
Another possibility would be to create a variable to remember the last activated subwindow.

Avec112, 2016-09-12:
I am not using Tab mode.

r-a-v-a-s, 2016-09-23:
Let me know if it crashes with 2.1.3

closed 2016-09-23 by r-a-v-a-s