LibreCAD/LibreCAD #815: Lines in construction layer disapear when out of viewport

damienflament, 2016-09-10:
When a line in a construction layer goes out of the viewport (by zoom panning or changing the zoom level), it disapear.

Steps to reproduce
--------------------
1. Open a new document,
2. On the default layer "0", create a line (using 2 points line, Vertical or Horizontal tool) starting on a Meta Grid guide and ending on another,
3. Modify layer attributes to make it a Construction Layer
    => the line is displayed as an infinite line,
4. Make the previously drawn line going out of the viewport using Zoom Panning or by changing the zoom level (use the Meta Grid guides previouly choosen as marks to help you) but keep its continuation visible.

Expected result
----------------
The line is visible in the viewport.

Actual result
-------------
The line disapears.

LibreCAD information
----------------------
- Version: 2.1.0
- Compiler: GNU GCC 6.1.1
- Compiled on: Jun 26 2016
- Qt Version: 5.7.0
- Boost Version: 1.60.0

System information
--------------------
- OS: ArchLinux
- Linux Kernel: 4.7.2

Possible explanation
---------------------
It seems obvious that the LibreCAD kernel considers the base segment to choose when to hide it but
do not take care of its infinite property (that of its layer).

r-a-v-a-s, 2016-09-10:
I appreciate the excellent bug report;
however, this has already been fixed.
https://github.com/LibreCAD/LibreCAD/commit/c13a55cb87f17f297f813775b2388d1e2b43618e

Technically you're using 2.1.1 (when it was introduced);
an oversight causes it to show as 2.1.0.
I'll work on releasing 2.1.2 this coming week.

damienflament, 2016-09-10:
I searched within issues using "construction" keyword and didn't find it...
This coming week ! Thanks.
I need to ensure package maintainer in ArchLinux team take care of the update as soon as possible. Because this bug is really annoying when you use a lot the construction lines !

Thanks.

kfoltman, 2016-09-10:
@r-a-v-a-s can you hold a little? I have an almost-done patch for some undo issues that currently make polylines very difficult to use, but I got stuck with that changeAttribute block edit problem.

I can commit what I have now, just to fix the worst part of the issue perhaps.

r-a-v-a-s, 2016-09-10:
Take your time and just add it to the master branch for now.
If I can find the motivation, 2.2.0 could be ready soon enough.

kfoltman, 2016-09-10:
Sure, I just think a release with those undo bugs could annoy everyone who uses polylines and undo - in current git master undoing things like add node, moving nodes etc. doesn't work at all, and in some workflows (CAM etc.) polylines are quite common. Which is why I think a fix that addresses that, but not the changeAttribute stuff, is still a possible improvement.

damienflament, 2016-09-10:
This software is clean and seems well maintained. Source code also.
I don't have time today. But one day, if I still use it, maybe can I help developing ?

Thanks a lot.

r-a-v-a-s, 2016-09-10:
@kfoltman I'm going to cancel my remaining projects, as 2.2.0 already has plenty of changes.
Therefore, we only need to wait for your changes and a week or two for a beta phase.

@damienflament Considering I put in a substantial amount of effort over past year 
cleaning various aspects, I appreciate you saying that.
Although, there is plenty left to clean... and we really do need any help we can get.
The first thing you could do is write an Arch Linux section for:
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

damienflament, 2016-09-11:
In ArchLinux, we use ArchLinux User Repository to build from source.
There is a PKGBUILD to build LibreCAD from Git : [librecad-git](https://aur.archlinux.org/packages/librecad-git/).
But it has not be maintained for 1 year. Generally, I prefer to use distribution package, for stability. So, I didn't even try the AUR build.
So, the first thing I can do is to maintain the PKGBUILD and use it ! 
Then, I can update LibreCAD references in the ArchLinux Wiki to mention the AUR package and add a sentence like "is a well maintained software" or "received updates very often".

I think I shoud stop spamming this bug ticket with unrelated things !

r-a-v-a-s, 2016-09-17:
https://github.com/LibreCAD/LibreCAD/releases/tag/2.1.2

closed 2016-09-17 by r-a-v-a-s