LibreCAD/LibreCAD #816: Crash in plugin: Ubuntu 16.04/Branch 2.1

Xia-Weiwen, 2016-09-13:
Hi, there.
If I use ```Document_Interface::getEnt``` and then ```Plug_Entity::getData```, the program crashes quite often (not every time).
When debugging, I found that when crash happens, ```Document_Interface::getEnt``` returns a pointer pointing to 0xc0, which is not a valid Plug_Entity pointer. The plugin called *Same Property* uses such functions and it causes crash as well.
It seems that the reason is related to the system and the LibreCAD version.
If I use **LibreCAD 2.1 branch, Ubuntu 16.04 64bit, Qt 5.3.2, boost 1.58.0**, crash happens (but not every time). Note: I have tried on a desktop and a laptop pc (both in Ubuntu 16.04).
But under the following conditions, crash does not happen (at least I didn't see):
1. LibreCAD master branch, Ubuntu 16.04 64bit, Qt 5.3.2
2. LibreCAD 2.1 branch, Windows 7 64bit/32bit, Qt 5.6/Qt 5.7
3. LibreCAD 2.1 branch, Lubuntu 14.04
Maybe it's about the operating system or some libraries LibreCAD uses?
Could you please help solve this?

Xia-Weiwen, 2016-09-14:
Now I'm seeing the same problem using master branch in Ubuntu 16.04.
I'm sure the 2.1 branch I tested is the newest version.

r-a-v-a-s, 2016-09-22:
I couldn't reproduce it on OS X with clang.
It crashed on the second try on OpenSUSE / lxqt with gcc 6.2.1
Both computers have Qt 5.6.1

r-a-v-a-s, 2016-09-22:
I seems like `getEnt()` returns a valid pointer,
and `Plug_Entity::entity` is the `0xc0`.
I don't know what that means and I couldn't find a solution.

Xia-Weiwen, 2016-09-23:
I've seen the same thing.
And now on Windows 7 (64 bit) with Qt 5.6/5.7 (32bit MinGW), I see crash happen in the same way but only in debug mode.

Xia-Weiwen, 2016-09-25:
Something new:
Activating **any** plugin causes crash if release mode selected.

Evironment:
+ OS: Windows 10 Home Edition
+ LibreCAD Version: 2.2.0-alpha (master branch)
+ Compiler： GNU GCC 5.3
+ Qt Version：5.7.0
+ Boost Version: 1.61.0

Crash position:
```
\librecad\src\main\qc_applicationwindow.cpp:459
```
Warning in log:
```
QObject: Cannot create children for a parent that is in a different thread.
(Parent is LC_Align(0x28c3cbe8), parent's thread is QThread(0x28c3c3a8), current thread is QThread(0x263760e8)
QG_LayerWidget::update: nullptr lastLayer
```

r-a-v-a-s, 2016-09-27:
That seems like a separate Windows 10 / Qt 5.7 issue.
I never encountered that on any system.

If you want to write plugins,
and want to make a significant contribution to LibreCAD,
I encourage you to consider:
https://github.com/LibreCAD/LibreCAD/issues/822

Continuing to patch the current system,
and allowing people to write plugins with it,
is like fostering a disease that becomes
harder and harder to cure.

Xia-Weiwen, 2016-09-30:
Thanks.
I'll try.

Xia-Weiwen, 2016-10-02:
I found something:
Adding ``` : public Plug_Entity``` to the end of ```/librecad/src/main/doc_plugin_interface.h : 50``` works.
I'll keeping studying this.

fa201, 2020-04-13:
I could not reproduce the issue.
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

mi-hol, 2022-06-07:
should this issue be closed?

Xia-Weiwen, 2022-06-10:
Yes. I think so. It's been too long.

closed 2022-06-10 by Xia-Weiwen